STMicroelectronics/STM32CubeH7 #75: No validity chekcing on dev_desc->bMaxPacketSize

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void  USBH_ParseDevDesc(USBH_DevDescTypeDef *dev_desc, uint8_t *buf, uint16_t length)

- Location:
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L355

- Type:
	- Denial-of-Service.

- Result:
  - A malformed USB device packet may cause the system hang when it tries to communicate with the outside world.

- Description:
	- The function USBH_ParseDevDesc() parses the device descriptor by input data from a USB device. 
    - The valid max packet size of the device descriptor should be 8, 16, 32, and 64 as USB specification required. However, the function USBH_ParseDevDesc() doesn’t check the value of dev_desc->bMaxPacketSize as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L355. The variable dev_desc->bMaxPacketSize will be used as the size to construct the control pipe between host and device as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c#L828. If bMaxPacketSize is zero, the firmware will get the error status USBH_FAIL in the function USBH_HandleControl() called by the function USBH_CtlReq() when trying to communicate with the outside world by IN and OUT pipe in the future and the host will try to re-enumerate. This process will loop again and again.

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a usb disk

3. Use the attached Bug1.txt to replace USB device packet

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.
- To patch it, the program should check if dev_desc->bMaxPacketSize is equal to 8, 16, 32 or 64. At least, it should be greater than zero.
[Bug1.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381197/Bug1.txt)

ALABSTM, 2020-11-24:
Hi @TheSilentDawn,

Thank you for your interest in our products and software solutions. Thank you also for this report and for all the other ones. They will be transmitted to our development teams for analysis.

However, may I first ask you whether these cases you are reporting (or at least some of them) are real error cases you faced while using our library or simulated test cases you designed based on a code review? Thank you in advance for your answer.

With regards,

TheSilentDawn, 2020-11-25:
Hi, @ALABSTM,
All the vulnerabilities reported by me are discovered by the research work of our team. We are building an automatic method to find the bugs. And all the testcases are checked manually before reporting to make sure it could be leveraged by the attackers. Also, these vulnerabilities could be patched as described in the **Addition context** part. Could you please share an email? When our paper is ready for submission, I will let you know and share the draft with you asap.

ALABSTM, 2020-12-02:
Hi @TheSilentDawn,

Thank you for your contribution. All the reports you sent will be forwarded to our development teams. I will get back to you as soon as they provide me with their feedback.

Thank you again for your contribution and thank you in advance for your patience.

With regards,

ALABSTM, 2021-01-18:
Hi @TheSilentDawn,

I hope you are doing well. Our technical committee discussed the several points you reported. Actions will be taken to make the necessary updates. I will keep you informed.

Any news or progress from your side? Thank you for your answer and thank you once more for your contribution.

With regards,

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

TheSilentDawn, 2021-05-31:
@ALABSTM 
Hello Ali LABBENE,

I'm a Ph.D. student from the University of Chinese Academy of Sciences and the University of Georgia. We are working on a fuzzing tool for automatic bug discovery. In the past year, I have reported several bugs that influence the MCU product line of STMicroelectronics. Some other STMicroelectronics engineers and your team have confirmed with us and the reported bugs have been patched. Currently, we are working on a research paper that systematically describes our new method. I wonder if your team can help apply for CVE IDs for these bugs so that we can refer to these CVE IDs and state our responsible disclosure with confidence.

Thanks for your help. If you want, we can also send you a draft of our paper before submission so that you can check whether the wording is appropriate. Below is a list of relevant bugs we reported.

https://community.st.com/s/question/0D53W00000BKF70SAH/stmplc-bug1
https://community.st.com/s/question/0D53W00000BKF7PSAX/stmplc-bug2
https://community.st.com/s/question/0D53W00000BKF7QSAX/stmplc-bug3
https://community.st.com/s/question/0D53W00000BKF8rSAH/stmplc-bug4
https://community.st.com/s/question/0D53W00000BKF8sSAH/stmplc-bug5
https://community.st.com/s/question/0D53W00000BKF9uSAH/stmplc-bug6
https://community.st.com/s/question/0D53W00000BKGAMSA5/stmplc-bug7
https://community.st.com/s/question/0D53W00000BKGANSA5/stmplc-bug8
https://community.st.com/s/question/0D53W00000BKGJWSA5/stmplc-bug9
https://github.com/STMicroelectronics/STM32CubeH7/issues/75
https://github.com/STMicroelectronics/STM32CubeH7/issues/76
https://github.com/STMicroelectronics/STM32CubeH7/issues/77
https://github.com/STMicroelectronics/STM32CubeH7/issues/78
https://github.com/STMicroelectronics/STM32CubeH7/issues/79
https://github.com/STMicroelectronics/STM32CubeH7/issues/80
https://github.com/STMicroelectronics/STM32CubeH7/issues/81
https://github.com/STMicroelectronics/STM32CubeH7/issues/82
https://github.com/STMicroelectronics/STM32CubeH7/issues/83
https://github.com/STMicroelectronics/STM32CubeH7/issues/84

We look forwards to your reply.

Sincerely,
Wenqiang Li
Email: wenqiang-li@outlook.com, liwenqiang@iie.ac.cn

TheSilentDawn, 2021-06-22:
Hi @ALABSTM ,
Thanks for your help.

CHAMSTM, 2021-11-15:
> Hi @ALABSTM How about the CVE requesting process? Our research paper needs CVE ID support. And we find more bugs in the STM32 MCU package. Should we report it here or email them to you?

Hello,
Our Security Support team is trying to create CVE IDs and will share them asap.

Kind Regards,

CHAMSTM, 2021-11-15:
Issue fixed in USB Host V3.4.0