STMicroelectronics/STM32CubeH7 #76: No validity checking on the variable cfg_desc->wTotalLength

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void USBH_ParseCfgDesc(USBH_CfgDescTypeDef *cfg_desc, uint8_t *buf, uint16_t length)

- Location:
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L395

- Type:
	- Buffer Overflow

- Result:
	- The system could be configured incorrectly with wrong parameters.  

- Description:
	- The function USBH_ParseCfgDesc() parses the configuration descriptor, interface descriptor, and endpoint descriptor by input data from a USB device. 
    - However, it doesn’t check the validity of the variable cfg_desc->wTotalLength compared with the total length of the input buffer as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L395. This will cause the following program including calling to the function USBH_GetNextDesc(), USBH_ParseInterfaceDesc() and USBH_ParseEPDesc() configure the system incorrectly.

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a usb disk

3. Use the attached Bug2.txt to replace USB device packet

**Additional context**
- To patch it, the program should check if reach the end of input buffer when plus cfg_desc->wTotalLength.
[Bug2.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381225/Bug2.txt)

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

CHAMSTM, 2021-07-26:
Already fixed in USBH V3.4.0

closed 2022-03-14 by ALABSTM