STMicroelectronics/STM32CubeH7 #77: No validity checking on the variable cfg_desc->bLength

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void USBH_ParseCfgDesc(USBH_CfgDescTypeDef *cfg_desc, uint8_t *buf, uint16_t length)

- Location:
    - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L393

- Type:
	- Buffer Overflow.

- Result:
	- The system could be configured incorrectly with the 
[Bug3.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381244/Bug3.txt)
wrong parameters.

- Description:
	- The function USBH_ParseCfgDesc() parses the configuration descriptor, interface descriptor, and endpoint descriptor by input data from a USB device. 
  - However, it doesn’t check the validity of the variable cfg_desc->bLength compared with the total length of the input buffer and the restriction of enumeration requirement by USB specification as shown in  https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L393. This will cause the following function USBH_GetNextDesc() and others access wrong memory region and the system will be configured incorrectly.

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug3.txt to replace the USB device packet

**Additional context**
- To patch it, the program should check if reach the end of the input buffer when plus cfg_desc->bLength.

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

CHAMSTM, 2021-11-15:
Fixed in USB Host V3.4.0

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,

closed 2022-03-14 by ALABSTM