STMicroelectronics/STM32CubeH7 #78: Lack hardware wake-up support checking

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void USBH_ParseCfgDesc(USBH_CfgDescTypeDef *cfg_desc, uint8_t *buf, uint16_t length)

- Location:
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L399

- Type:
	- Denial-of-Service.

- Result:
	- The system will hang when trying to set a remote wake-up feature.

- Description:
	- The function USBH_ParseCfgDesc() parses the configuration descriptor, interface descriptor, and endpoint descriptor by input data from a USB device. 
    - And it set the variable cfg_desc->bmAttributes by the input data from the USB device. This variable will be used as part of a judgment in the function USBH_Process() as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c#L643. With a malformed value, the remote wakeup may be enabled as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c#L643. If the hardware doesn’t support this feature, the system will hang due to a FAIL return value by the function USBH_HandleControl(). 

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug4.txt to replace the USB device packet

**Additional context**
- To patch it, the program should check if the hardware supports a remote wake-up
[Bug4.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381270/Bug4.txt)
 feature.

ALABSTM, 2020-11-24:
Hi @TheSilentDawn,

Thank you for this other report.  If I understood the idea we should ensure that the software embedded within the *host* checks whether the value *bmAttributes* provided by the *device* is correct? Which reference should the *host* base itself on to perform such a check? Should we not assume the data provided by the *device* is correct?

Otherwise, when to stop checking? In case the *host* has a reference to base itself on, then would it not also make sense to check the validity of this reference too?

I hope my questions make sense. My goal is to be sure I understood correctly your point before reporting it to our development teams. I am looking forward to reading your reply. Thank you again.

With regards,

TheSilentDawn, 2020-11-25:
Hi @ALABSTM,
No, I think the software embedded within the host should check if the hardware or the board supports or enables the wake-up feature. If not, the firmware shouldn't check _bmAttributes_  as the current logic which will lead the whole system hanging. And to patch it, the wake-up feature checking codes should be added before the _bmAttributes_ is accessed in the function and position reported. If anything not clear, please let me know. Thanks for your help.

ALABSTM, 2020-12-02:
Hi @TheSilentDawn,

Please allow me this couple of questions:
* Is there any way for the host to check whether the device supports the wake-up feature other than the `bmAttributes` read from the device descriptor?
* You mentioned *"the wake-up feature checking codes"*. Are you referring to some piece of code already present in our source files that should be moved before the reading of the `bmAttributes`?

Thank you,

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,