STMicroelectronics/STM32CubeH7 #80: No checking on both IN and OUT pipe constructed

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static USBH_StatusTypeDef USBH_MSC_InterfaceInit(USBH_HandleTypeDef *phost)

- Location:
	- From line 180 to line 200 in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc.c

- Type:
	- Denial-of-Service.

- Result:
	- The system will hang when try to communicate with the endpoint.

- Description:
	- The function USBH_MSC_InterfaceInit() inits the status of MSC handler. It initializes the IN endpoint and OUT endpoint as shown from line 180 to line 200 in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc.c. 
    - However, when the variable bEndpointAddress of endpoint descriptor are both masked as IN or OUT without checking as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L468, the MSC handler will also only initialize the IN or OUT part as shown from line 180 to line 200 in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc.c.  

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug6.txt to replace the USB device packet. 
[Bug6.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381333/Bug6.txt)

**Additional context**
- To patch it, the program should check both IN and OUT pipe is constructed.

ALABSTM, 2020-11-24:
Hi @TheSilentDawn,

Thank you for this other report. To be sure I understood your point, please allow me these questions:
* We are dealing with the case of a **simultaneously** IN and OUT endpoint? This is what I understood from the expression *"the variable bEndpointAddress of endpoint descriptor are both masked as IN or OUT"*.
* We are requesting from the software to check both? This is what I understood from the expression *"the program should check **both** IN and OUT pipe is constructed"*

Thank you in advance for your clarifications.

With regards,

TheSilentDawn, 2020-11-25:
Hi @ALABSTM,
Yes, the firmware should open an IN pipe and an OUT pipe to communicate with a device normally. However, as the report described, the driver codes do not check this. It will open two IN pipes or two OUT pipes unexpectedly which will block the communication in the subsequent execution of the firmware. If anything not clear, please let me know. Thanks for your help.

ALABSTM, 2020-12-02:
Hi @TheSilentDawn,

Thank you for the clarification. Hence, the case you are describing is an interface exposing 2x IN endpoints **or** 2x OUT enpoints **instead of** 1x IN endpoint and 1x OUT endpoint (which is the normal case for a **MSC**).

I will forward your report to our development teams. I will get back to you as soon as they provide me with their feedback.

With regards,

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,

closed 2022-03-14 by ALABSTM