STMicroelectronics/STM32CubeH7 #81: No checking if ep_descriptor->wMaxPacketSize is greater than zero

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void  USBH_ParseEPDesc(USBH_EpDescTypeDef  *ep_descriptor, uint8_t *buf)

- Location:
    - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L470

- Type:
	- Denial-of-Service.

- Result:
	- The system will hang when try to communicate with the endpoint.

- Description:
	- The function USBH_ParseEPDesc parses the endpoint descriptor of a USB device. 
    - It doesn’t check if the variable ep_descriptor->wMaxPacketSize is greater than zero as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L470. If zero, the MSC handler will not able to communicate with the outside world as shown from line 180 to line 200 in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc.c

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug7.txt to replace the USB device packet. 
[Bug7.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381343/Bug7.txt)

**Additional context**
- To patch it, the program should check if ep_descriptor->wMaxPacketSize is greater than zero.

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,

closed 2022-03-14 by ALABSTM