STMicroelectronics/STM32CubeH7 #82: No checking if ep_descriptor->bInterval is greater than zero

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
	- static void  USBH_ParseEPDesc(USBH_EpDescTypeDef  *ep_descriptor, uint8_t *buf)

- Location:
https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L471

- Type:
	- Denial-of-Service.

- Result:
	- The system will hang.

- Description:
	- The function USBH_ParseEPDesc parses the endpoint descriptor of a USB device. 
    - It doesn’t check if the variable ep_descriptor->bInterval is greater than zero. This variable will be used in the class to set the polling interval, for example as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Class/AUDIO/Src/usbh_audio.c#L858. If set zero, the system will hang when polling operation. 

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug8.txt to replace the USB device packet. 
[Bug8.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381346/Bug8.txt)

**Additional context**
- To patch it, the program should check if ep_descriptor->bInterval is greater than zero.

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,

closed 2022-03-14 by ALABSTM