STMicroelectronics/STM32CubeH7 #84: Buffer Overflow due to the variable ep_descriptor->bLength

TheSilentDawn, 2020-10-14:
**Describe the set-up**
- Software:
	- STM32Cube MCU & MPU Packages
- Version:
	- STM32Cube_FW_H7_V1.8.0
- Verification Hardware Platform:
	- STM32H7B3

**Describe the bug**
- Function:
    - static void  USBH_ParseEPDesc(USBH_EpDescTypeDef  *ep_descriptor, uint8_t *buf)

- Location:
    - https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L466

- Type:
	- Buffer Overflow

- Result:
	- The system could be configured incorrectly with wrong parameters.

- Description:
	- The function USBH_ParseEPDesc() parses endpoint descriptor.  It’s called by the function USBH_ParseCfgDesc() as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L423. 
    - It doesn’t check the validity of the variable ep_descriptor->bLength compared with the total length of the input buffer which may cause a buffer overflow by the following called function USBH_GetNextDesc() as shown in https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L419. 

**How To Reproduce**
1. Running MSC_Standalone application on the STM32H7B3I platform

2. Plug a USB disk

3. Use the attached Bug10.txt to replace the USB device packet. 
[Bug10.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/5381420/Bug10.txt)

**Additional context**
- To patch it, the program should check if reach the end of the input buffer when plus ep_descriptor->bLength.

ALABSTM, 2021-01-18:
ST Internal Reference: 99173

ALABSTM, 2021-06-21:
Hi @TheSilentDawn,

I hope you are doing well. This point you raised and all others have been fixed in the frame of **v3.4.0** of the USB Host Library found in the [stm32_mw_usb_host](https://github.com/STMicroelectronics/stm32_mw_usb_host) repository. Thank you again for your report.

With regards,

TheSilentDawn, 2021-06-21:
> Hi @TheSilentDawn,
>
> I hope you are doing well. This point you raised and all others have been fixed in the frame of **v3.4.0** of the USB Host Library found in the [...]

TheSilentDawn, 2021-06-21:
> Hi @TheSilentDawn,
>
> I hope you are doing well. This point you raised and all others have been fixed in the frame of **v3.4.0** of the USB Host Library found in the [...]

Hi @ALABSTM,
I'm glad to hear that all of the fixes have been issued. Actually, these bugs are found by an automatic fuzzing tool developed by my lab. We are now working on a research article that discusses our novel technique in detail. I'm wondering whether your team might assist us in obtaining CVE IDs for these issues so that we can confidently refer to these CVE IDs and indicate our responsible disclosure.

ALABSTM, 2021-06-28:
Hi @TheSilentDawn,

Your request to obtain CVE IDs has been forwarded to our teams. I will get back to you as soon as I have feedback.

With regards,

TheSilentDawn, 2021-07-05:
Hi @ALABSTM ,
Thanks for your help.

ALABSTM, 2022-03-14:
Hi @TheSilentDawn,

Hope you're fine. Just to inform you the fix has been published in the frame of **v1.10.0** release.

With regards,

closed 2022-03-14 by ALABSTM