STMicroelectronics/STM32CubeH7 #89: HAL_SDRAM_Init should take timing argument as const

gmarull, 2020-11-09:
**Describe the set-up**

STM32H747I-DISCO, using Zephyr RTOS.

**Describe the bug**

```
HAL_StatusTypeDef HAL_SDRAM_Init(SDRAM_HandleTypeDef *hsdram, FMC_SDRAM_TimingTypeDef *Timing)
```

Should ideally take `Timing` argument as `const`, i.e.

```
HAL_StatusTypeDef HAL_SDRAM_Init(SDRAM_HandleTypeDef *hsdram, const FMC_SDRAM_TimingTypeDef *Timing)
```

Ref.
https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_sdram.c#L171

Note that the same should happen for the LL counterpart, https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_fmc.c#L791

Other FMC functions can probably be adjusted.

**How To Reproduce**

Not causing any problem, just warnings if using a `const` structure.

**Additional context**

N/A

**Screenshots**

N/A

RKOUSTM, 2020-11-12:
Hi @gmarull ,

Thank you for this detailed report. This issue has also been previously reported by other users ([link](https://github.com/STMicroelectronics/STM32CubeF4/issues/10)).

This is one of the main points in the tasks queue of our development teams. We cannot share a date regarding its deployment and publication. We count on your patience and comprehension.

Thank you again for this report. 

With regards,

RKOUSTM, 2020-11-19:
Hi @gmarull, 

First, allow me to thank you for your contribution. The final decision of our technical committee is no plan to add a const inside the function. The reported point is already highlighted during the MISRA-C 2012 compliance rework (Rule-8.13). It has been decided, from the quality point of view, to derogate this rule as fixing it will break the compatibility with almost all HAL APIs and then update all examples and applications within the Cube deliverable and also on user application side.

Please, allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2020-11-19 by RKOUSTM