STMicroelectronics/STM32CubeH7 #90: Calling xQueueReceiveFromISR() from within Interrupt results in ASSERET trigger

rxa1031, 2020-11-14:
Hello,

Microcontroller in use: STM32H753BIT6

My code uses:
```
HAL_NVIC_SetPriority(USART1_IRQn, 5, 0);
HAL_NVIC_EnableIRQ(USART1_IRQn);
```

`Macro configASSERT is defined.`

My code calls  `xQueueReceiveFromISR()` inside the USART1_IRQ. I am observing issue with `xQueueReceiveFromISR()` (which is called with help of `osMessageQueueGet()`)

When I analyzed the issue it turned out that ASSERT was triggered at https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.c#L754

inside function `vPortValidateInterruptPriority()`.

I had to change lines:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.c#L72-L73

to:
```
#define portFIRST_USER_INTERRUPT_NUMBER       ( 6 )
#define portNVIC_IP_REGISTERS_OFFSET_16       ( 0xE000E3FA )
```

Also, I had to change line:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.c#L729

to:
`ucCurrentPriority = *(pcInterruptPriorityRegisters + ulCurrentInterrupt);`

To fix the ASSERT issue.

Please let me know your inputs.

Thanks,
@rxa1031

rxa1031, 2020-11-15:
ulCurrentInterrupt = 0x25

rxa1031, 2020-11-19:
Errata:
Code was using TIM2 IRQ and not UART1 IRQ
The actual problem is not the portNVIC_IP_REGISTERS_OFFSET_16 or portFIRST_USER_INTERRUPT_NUMBER values.

The issue exists because of 
`#define  TICK_INT_PRIORITY            ((uint32_t)0U) /*!< tick interrupt priority */`

And its use during configuring the Timer that is used by FreeRTOS.

_If `xQueueReceiveFromISR()` is called inside the FreeRTOS Timer interrupt the value copied to ucCurrentPriority (inside function vPortValidateInterruptPriority()) is ZERO._

I am collecting some more details with regards to _`HAL_InitTick()` which seems to be the culprit for FreeRTOS timer's priority setting_ and also might help me elaborate on #71 observations.

Regards,
@rxa1031

rxa1031, 2020-11-25:
Hello @ASELSTM ,

I observe concerns with FreeRTOS file GCC\ARM_CM4\port.c available with it. I am observing two ASSERTS / concerns in the file (triggered as a result of Reading from or Writing to a Queue from ISR):
 
It is observed that **pxCurrentTCB->pcTaskName == "IDLE"** for both the (software interrupt execution) cases:

**First Issue:**
Is observed for `configASSERT( ucCurrentPriority >= ucMaxSysCallPriority );` when variable uwTickPrio = TICK_INT_PRIORITY (either assigned directly or with help of  variable TickPriority), where macro TICK_INT_PRIORITY is assigned a value of 0.

If I _do not use a software interrupt_ to process the Queue, then because of uwTickPrio value (i.e. 0) being used as priority for FreeRTOS Timer (i.e. Timer 2), the ASSERT gets raised.

If I _use the software interrupt_, then I observe the _second issue_ stated below
 
**Second issue:**
Is observed for `configASSERT( ( portAIRCR_REG & portPRIORITY_GROUP_MASK ) <= ulMaxPRIGROUPValue );` especially when triggered software interrupts are being handled. The software interrupt is triggered by updating the related interrupt number NVIC->STIR

**Additional observations and concerns:**

1. May you please help me understand whether `TICK_INT_PRIORITY` value is user changeable?
2. Also should the `NVIC_SetPriorityGrouping(4)` or `NVIC_SetPriorityGrouping(0)` to be used? It is observed that `HAL_Init()` calls `HAL_NVIC_SetPriorityGrouping(4)`, hence if the `Priority Group is to be set to 0` then should `NVIC_SetPriorityGrouping(0)` called just after calling `HAL_Init()`?

**Flow:**
SW_FroTimer_IRQHandler() triggered -> SW_FroTimer_IRQHandler() serviced -> calls third party library function -> Third part library calls function rvAddEventToInterruptEvents() which I've coded -> rvAddEventToInterruptEvents() triggers another software interrupt -> SW_IRQHandler() executes for good number of times and then suddenly the issue 2 occurs.
Observation:  **pxCurrentTCB->pcTaskName == "IDLE"**

**My code has:**
// USART2 IRQ being used for software interrupt and is triggered from a third party library on which my code does not have any control. This execution is the result of execution of SW_FroTimer_IRQHandler() when reviewed in Call Stack.
#define SW_Int_IRQn			USART2_IRQn
#define SW_IRQHandler		USART2_IRQHandler

// This software interrupt is triggered from Timer 2 (which is used as the default FreeRTOS timer)
#define SW_FroTimer_Int_IRQn			SPI4_IRQn
#define SW_FroTimer_IRQHandler		SPI4_IRQHandler

below added to _main()_ function:
```			HAL_NVIC_SetPriority(SW_Int_IRQn, 6, 0);
			HAL_NVIC_EnableIRQ(SW_Int_IRQn);

			HAL_NVIC_SetPriority(SW_FroTimer_Int_IRQn, 6, 0);
			HAL_NVIC_EnableIRQ(SW_FroTimer_Int_IRQn);```

Inside `HAL_TIM_PeriodElapsedCallback()` below code is added:
```		if((htim->Instance == TIM2) && (nRTOSTimerAndFMCInit == nRTOSTimerAndFMCInitStatus))
		{
			CucHighMinusLowSystemTicks++;
			NVIC->STIR = SW_FroniusTimer_Int_IRQn;
		}```

The other piece of code that triggers the software interrupt is:
```void									rvAddEventToInterruptEvents
	(
		toTaskEventHandler	*		poTaskEventHandler
	)
{
	uint32_t u32IRQMasked;
	MvTraceTime;
	dvDisableInterrupts;
	{
		if (osOK != osMessageQueuePut(Q_EventHandlerHandle, &poTaskEventHandler, dDummyMessagePriority, 0 /* Do not use osWaitForever as IRQ is disabled */))
		{
 			// #DRA: #ToDo: add error handling here
			++u32EventHandlerErrorCounts;
			assert_failed((uint8_t*)__FILE__, __LINE__);
		}
		else
		{
			NVIC->STIR = SW_Int_IRQn; 
		}
	}
	dvEnableInterrupts;
	return;
}```

Is it possible for you to suggest some pointers on how to fix the issues.

Thanks
@rxa1031

rxa1031, 2020-11-27:
Hello @ASELSTM , @RKOUSTM ,

I request your views on below observations.

Based on all the analysis, I found out that the at least one of the two configASSERT() present with function vPortValidateInterruptPriority() will get triggered when the Queue related code executes from within a Software Interrupt ISR that is triggered by writing Interrupt number of the interrupt to be triggered to NVIC->STIR (example NVIC->STIR = SW_FroTimer_Int_IRQn;).

To me it appears that the workaround for now is to use **vTaskNotifyGiveFromISR()** and **ulTaskNotifyTake()** _along with the software interrupt_ (as shared below):

1. Create a handle FroTimerTaskHandle for task T_FronTimer() in main

2. SW_FroTimer_IRQHandler() / Software Inerrupt function to call vTaskNotifyGiveFromISR(), instead of executing the functionalities were executed with help of call to function rLowSystemCInterrupt(). i.e.

```
extern osThreadId_t FroTimerTaskHandle;

void SW_FroTimer_IRQHandler(void)
{
// Earlier function rLowSystemCInterrupt(); was called in place of below function
	vTaskNotifyGiveFromISR(FroTimerTaskHandle, NULL);
}
```
3. rLowSystemCInterrupt();  (which was earlier called from SW_FroTimer_IRQHandler()) is modified to call ulTaskNotifyTake(). i.e.
```
// The handle for this task is FroTimerTaskHandle
void T_FroTimer(void * argument)
{
	rLowSystemCInterrupt()
}

__INLINE void rLowSystemCInterrupt(void);
void rLowSystemCInterrupt(void)
{
	uint32_t u32IRQMasked;
	bool bFirstTick = true;
	while(true)
	{
		/* Loop until all timer ticks have been handled */
		while( bFroLibraryInialized )
		{
			uint32_t u32ItemCount = osMessageQueueGetCount(Q_CRC_FroTagsOutHandle);
			if(0 < u32ItemCount)
			{
				dvDisableInterrupts;
				{
					if (osOK != osMessageQueueGet(Q_CRC_FroTagsOutHandle, &poCRC_FroniusTagOut, NULL, 0 /* Do not use osWaitForever as IRQ is disabled */))
					{
						// #DRA: #ToDo: add error handling here
						// ++ Error Counts
						assert_failed((uint8_t *)__FILE__, __LINE__);
					}
				}
				dvEnableInterrupts;
			}
			else
			{
				break;
			}
		}
// Below line needed only when using function rLowSystemCInterrupt() is called from the task T_FroTimer()
		ulTaskNotifyTake( pdTRUE, portMAX_DELAY);
	}
}
```

Regards,
Rajeev

RKOUSTM, 2020-12-25:
Hi @rxa1031, 

Thank you for your contribution. Unfortunately we don't treat aspect related to Third_Party files in our GitHub repositories. As these files are out of the scope of STMicroelectronics, any modifications concerning ``port.c`` file are not implemented by our technical team. All aspects related to FreeRTOS are not handled in our GitHub repositories. They are rather treated at [FreeRTOS GitHub](https://github.com/freertos). 

Please allow me to close this issue now. Thank you again for your contribution.

With regards,

closed 2021-01-12 by RKOUSTM