STMicroelectronics/STM32CubeH7 #91: TIM15 is missing in macro IS_TIM_SLAVE_INSTANCE

mcdewitz, 2020-11-16:
**Describe the set-up**
* I wanted to use TIM15 in slave mode on the NUCLEO-H743ZI2 board (featuring the STM32H743ZIT6 rev. V)
* Init code for TIM15 is generated by CubeIDE 1.4.0, the MCU Package for H7 used is version 1.8.0

**Describe the bug**
 *  runs into a failed assertion when it comes to calling the function HAL_TIM_SlaveConfigSynchro. Therein the macro IS_TIM_SLAVE_MODE checks the Timer Instance features the Slave Mode Control Register SMCR
* The actual bug is that TIM15 is missing in the macro hence the assertion will fail.

**How To Reproduce**
To reproduce the bug USE_FULL_ASSERT needs to be defined. Running the following code should yield the aforementioned error:
`TIM_HandleTypeDef htim15;`
`TIM_SlaveConfigTypeDef sSlaveConfig = { 0 };`
`htim15.Instance = TIM15;`
`HAL_TIM_SlaveConfigSynchro(&htim15, &sSlaveConfig);`

**Additional context**
Bug is fixed for me by extending the macro as printed below. The TIM15 is working as I would expect with regards to the Ref Manual.
`/****** TIM Instances : Slave mode available (TIMx_SMCR.TS available )*********/`
`#define IS_TIM_SLAVE_INSTANCE(INSTANCE)   (((INSTANCE) == TIM1)   || \`
`                                           ((INSTANCE) == TIM2)   || \`
`                                           ((INSTANCE) == TIM3)   || \`
`                                           ((INSTANCE) == TIM4)   || \`
`                                           ((INSTANCE) == TIM5)   || \`
`                                           ((INSTANCE) == TIM8)   || \`
`                                           ((INSTANCE) == TIM12)  || \`
`					   ((INSTANCE) == TIM15))`

Oddly enough some other header files (only speaking for H7 series) indeed do contain TIM15 in the macro definition (e.g. stm32h730.h). To me it wouldn't be much of a surprise if the same problem exists in even more files accross the STM32 series, assuming the "silicon" for TIM15 is by and large the same.

ASELSTM, 2020-11-18:
Hi @mcdewitz,

Thank you for this report. Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2021-02-15:
Hi @mcdewitz,

The issue you pointed out has been confirmed, a fix will be implemented and made available in a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-02-15:
ST Internal Reference: 101414

ALABSTM, 2021-12-16:
Hi @mcdewitz,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-16 by ALABSTM

ALABSTM, 2021-12-16:
Related to #114.