STMicroelectronics/STM32CubeH7 #93: LL_SYSCFG_ETH_RMII definition wrong.

basvloon, 2020-11-18:
Hello,

while debugging RMII functionality I've discovered a bug in the stm32h7xx_ll_system.h header. It sets all the bits (including 'reserved' bits) instead of just the bits necessary to enable RMII mode.

Patch looks like this:
`@@ -130,7 +135,7 @@
 * @{
 */
 #define LL_SYSCFG_ETH_MII               0x00000000U                           /*!< ETH Media MII interface */
-#define LL_SYSCFG_ETH_RMII              SYSCFG_PMCR_EPIS_SEL                   /*!< ETH Media RMII interface */
+#define LL_SYSCFG_ETH_RMII              SYSCFG_PMCR_EPIS_SEL_2                /*!< ETH Media RMII interface */
 /**
   * @}
   */`

Hope this helps.

Best Regards,
Bas

ASELSTM, 2020-11-18:
Hi @basvloon, 

Thank you for this report. Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-11-19:
Hi @basvloon,

Our development teams confirmed the issue you pointed out. They also agreed on the fix you have suggested, it will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-11-19:
ST Internal Reference: 96616