STMicroelectronics/STM32CubeH7 #95: STM32H7: Ethernet TX callback complete not called intermittently

emillindq, 2020-11-23:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * STM32H747I DISCO
 * gcc version 9.2.0 (zephyr-crosstool-NG 1.24.0.4)
* This HAL version 1.8.0

**Describe the bug**
Please refer to https://github.com/zephyrproject-rtos/zephyr/issues/29915 and https://github.com/zephyrproject-rtos/zephyr/pull/29944
To summarize, TxCpltCallback is not called all the time, and it appears to be related to some incomplete memory transactions. A fix is described in provided links: adding `__DSB()` to a certain line in this HAL and before/after the `HAL_ETH_Transmit_IT` resolves it, depending on compiler optimization level. *However*, adding the line in this HAL works in all cases.

**How To Reproduce**
1. Zephyr ethernet driver waits for a semaphore from TxCpltCallback, but it never comes as the callback is never called. So it times out.

2. Ethernet driver for stm32h7xx (stm32h7xx_hal_eth.c)

3. Just transmit an ethernet package

4. Follow the scenario in the provided links.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RKOUSTM, 2020-12-01:
Hi @emillindq,

Thank you for your contribution. Your report will be transmitted to our development teams to have this update in the frame of a future release. Thank you once more for this contribution.

With regards

emillindq, 2021-05-28:
https://github.com/zephyrproject-rtos/zephyr/pull/30403 fixes this, and so I close this issue

closed 2021-05-28 by emillindq