STMicroelectronics/STM32CubeH7 #96: No means to detect ASSERT trigger without a debugger for configASSERT(0)

rxa1031, 2020-11-27:
Hello @RKOUSTM ,

Please refer to below line of code: https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/include/FreeRTOSConfig_template.h#L149

If `configASSERT(0)` gets executed, then without a debugger it is impossible to know why code got stuck.

May you please add a weak function to the statement, which can later be coded by the user.

Is it possible to change the line of code to:

`#define configASSERT( x ) if ((x) == 0) {taskDISABLE_INTERRUPTS(); rvUserAssertInformFunction(); for( ;; );}`

Please share your views.

Thanks,
Rajeev

RKOUSTM, 2020-12-01:
Hi @rxa1031,
 
Thank you for this report. Unfortunately we don't treat aspect related to FreeRTOS in our GitHub repositories. As the Third_Party files are out of the scope of STMicroelectronics, any modification concerning these files is not implemented by our technical team. What about reporting this issue to the [FreeRTOS GitHub](https://github.com/freertos)?  
 
Please allow me to close this issue now. Thank you again for your contribution. 

With regards,

rxa1031, 2020-12-01:
Hello @RKOUSTM ,

The file STM32CubeH7/Middlewares/Third_Party/FreeRTOS/Source/include/FreeRTOSConfig_template.h is provided in STM32CubeH7 Repository, and hence I raised the concern here.

Also the file **FreeRTOSConfig.h** is available with many examples which are being provided within the STM32CubeMX repository. example:

https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Projects/STM32H750B-DK/Demonstrations/STemWin/Config/FreeRTOSConfig.h#L164

Hence, I have raised the concern. Also it was earlier informed that ST uses customized version of FreeRTOS. I believe that the issue I am reporting must be addressed by STM32CubeH7 developers.

Regards,
Rajeev

rxa1031, 2020-12-01:
Please review various **STM32CubeH7** provided various examples that include **FreeRTOSConfig.h** file. Please check the declaration of macro **configASSERT( x )** in above informed file(s).

RKOUSTM, 2020-12-01:
Hi @rxa1031,

ST proposes the usage of FreeRTOS in the context of the STM32Cube ecosystem. Any problem related to ``Third_Party`` files itself, we suggest to contact FreeRTOS supplier as any modification or update in these files is not implemented by our technical team. 

Please allow me to close this issue now. Thank you again for your report.

With regards,

rxa1031, 2020-12-01:
Hello @RKOUSTM , @ALABSTM 

Thanks for the information. I assume you mean to inform that the **STM32CubeH7** community / github page takes core of the **STM32Cube MCU Package for STM32H7 Series** only.

I am not aware which ST community takes care of integrating the FreeRTOS. Some ST community modified the FreeRTOS code when porting it for ST Microcontrollers. Example: Addition of `enum osPriority_t` which contains the value `osPriorityISR`. The enum is now used even by STM32CubeH7 repository.

May you please help me with contact (preferably email) or @ for the community who is involved with integrating the RTOS.

To me it appears that @ALABSTM added the file FreeRTOSConfig.h to the STM32CubeH7 repository.

I am not sure how the process works at ST, and which Enhancements (and bugs) to be reported to STM32CubeH7 community and which of them to be reported to STM32CubeMX community,

I plan to report this issue on STM32CubeMX ST community web page, probably they can help add the requested implementation.

Thanks,
Rajeev

RKOUSTM, 2020-12-07:
Hi @rxa1031,

Thank you for this report, once again. However, this issue is out of the scope of STMicroelectronics' activity as the concerned piece of software is a third party's one. As mentioned in a previous comment, aspects related to FreeRTOS are not handled in our GitHub repositories. They are rather treated at [FreeRTOS GitHub](https://github.com/freertos). 

Indeed, all aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at ST Community level, precisely within the [Cube MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section.

Thank you for your comprehension.

With regards,

closed 2020-12-07 by rxa1031