STMicroelectronics/STM32CubeH7 #99: LL_USART_ClockInit issue.

basvloon, 2020-11-30:
Hello,

According to the 'slave' functionality of the USART IP you can set the clock phase and polarity, which I would expect. In LL_USART_ClockInit there is no implementation to set the clock phase and polarity in case CLK_EN is off, which is should be done and is valid valid in case the IP is used in slave mode. At least according to the RM.

Quote from RM: When the peripheral is used in SPI slave mode, the frequency of peripheral clock source
(usart_ker_ck_pres) must be greater than 3 times the CK input frequency.
The CPOL bit and the CPHA bit in the USART_CR2 register are used to select the clock
polarity and the phase of the external clock, respectively (see Figure 583).
An underrun error flag is available in slave transmission mode. This flag is set when the first
clock pulse for data transmission appears while the software has not yet loaded any value to
USART_TDR.
The slave supports the hardware and software NSS management.

Am I wrong or is this not correctly implemented ?

Thanks for the feedback.

Best Regards,
Bas

ASELSTM, 2020-12-10:
Hi @basvloon,

First we would like to thank you for your contribution. 

No, you are not wrong. Indeed, the actual implementation of the LL_USART_ClockInit() is incorrect and won't allow the clock phase and the clock polarity configuration when CR2_CLKEN is cleared. 

A fix will be implemented  and will made available in the frame of a next release. 

Best regards,

ASELSTM, 2020-12-10:
ST Internal Reference: 98043