STMicroelectronics/STM32CubeH7 #100: CMSIS_OS2.C: Call to enable critical section missing before a call to any Function that ends with FromISR.

rxa1031, 2020-12-03:
Hello,

I am observing that the **cmsis_os2.c** file does not call make a call to `taskENTER_CRITICAL()` before if executes a function XXXFomISR() and also does not execute `taskEXIT_CRITICAL()` after funtion XXXFomISR() gets executed.

https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1099
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1121
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1142
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1562
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1667
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1708
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1769

I assume the call to critical section enable and disable should be made when executing a function that acts upon queue, semaphore, or time, irrespective of whether the function is called from within a task (like `xQueueReceive()`) or from within an ISR (like `xQueueReceiveFromISR()`).

Please share your views.

Thanks,
Rajeev

rxa1031, 2020-12-03:
Please review: Light weight example 3 – introducing preemption and interrupt safety

Available at location https://www.freertos.org/message_passing_performance.html

RKOUSTM, 2020-12-09:
Hi @rxa1031,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer.

Thank you for your patience and thank you again for your contribution.

With regards,

rxa1031, 2021-03-23:
Hello @ALABSTM ,

Are there any findings / updates that you may be able to share with me.

Thanks,
Rajeev

RKOUSTM, 2021-08-04:
Hi @rxa1031, 

I hope you are fine. Our development teams said your proposal could not be taken into consideration for the following reason:

> The xxxFromISR() functions are meant to be used exclusively within an interrupt context. Their use outside the interrupt context is not advised . But if a user decides to use them outside interrupt context, he has obviously to make the choice of protecting or not through critical sections. On the CMSIS_OS side, we cannot implement a change that is not generic and related to a non-recommended use case.

 Please allow me to close this issue now and thank you for your comprehension. We are looking forward to reading from you again.

With regards,

closed 2021-08-04 by RKOUSTM