STMicroelectronics/STM32CubeH7 #101: RCC_AHB2RSTR register for DCMI resetting on STM32H743VI

othmaneAZ, 2020-12-06:
**Caution**
This not a bug report, I'm just trying to understand. 

**Describe the set-up**
 * The board : Custom board based on the STM32h743VI MCU.
 * Writing device tree while trying to port this custom board on Zephyr

**Additional context**
Hello, In the Application Note of the DCMI interface, it is said that It is recommended to reset the DCMI peripheral and the camera module before starting the configuration. The DCMI can be reset by setting the corresponding bit in the RCC_AHB2RSTR register, which resets the clock domains. Thought, I can't find the corresponding bit in the RCC_AHB2RSTR of the STM32H743VI MCU. Is it normal ?

RKOUSTM, 2020-12-08:
Hi @othmaneAZ,

Thank you for your contribution. According to [RM0433](https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/c9/a3/76/fa/55/46/45/fa/DM00314099/files/DM00314099.pdf/jcr:content/translations/en.DM00314099.pdf), the digital camera interface ``(DCMI)`` can be reset by setting the camera interface bit ``(Bit 0 CAMITFRST)`` in the ``RCC_AHB2RSTR`` register. Indeed, an update in the reference manual will be proposed to rename the ``CAMITFRST`` bit by the ``DCMIRST`` and will be available soon.

Thank you again for your contribution. 

With regards,

RKOUSTM, 2020-12-08:
ST Internal Reference: 97861

othmaneAZ, 2020-12-08:
Thank you for your answer, this update will for sure make things clearer. 

I have another question aside though, my custom board contains an OV77xx camera sensor and I'm trying to define it in my device tree as a node, but I don't know where to find the address of this device. I appreciate your help.

Best regards,

RKOUSTM, 2020-12-09:
Hi @othmaneAZ,

Thank you for your contribution. You can adapt [this](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H747I-DISCO/Examples/BSP) BSP example on STM32H747I-DISCO to drive the camera module  ``ov77xx`` for any other development board as ``STM32H734VI`` device. 

Thank you again for your contribution.

With regards,

RKOUSTM, 2022-02-23:
Hi @othmaneAZ,

In order to change all references from CAMITF to DCMI, a new version of RM0399 and RM0433 will be available as soon as possible on st.com site.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem document, please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

closed 2022-02-23 by RKOUSTM