STMicroelectronics/STM32CubeH7 #102: Use Boot -> APP： HAL_ Delay() will be stuck in APP, But it is normal in Boot

fyyxxm, 2020-12-10:
/ * Issue Description */ 
Boot -> APP： HAL_ Delay() will be stuck in APP, But it is normal in Boot.

In addition, it is normal （LL_mDelay()）to use LL Library in version 1.7.0 Pakeage 。

**Tools**
MCU:  STM32H743II
IDE:     KEIL 5.30
MCU Pakeage for STM32H7 Series: 1.8.0

/* JumpToApp Code */

/* Private typedef -----------------------------------------------------------*/
typedef void (*pFunction)(void);
/* Private define ------------------------------------------------------------*/
/* Private macro -------------------------------------------------------------*/
/* Private variables ---------------------------------------------------------*/
pFunction JumpToApplication;

void BSP_JumpToApp(void)
{
    /* Disable CPU L1 cache before jumping to the QSPI code execution */
    BSP_CPUCacheDisable();
    /* Disable Systick interrupt */
    BSP_Disable_IRQ();
    
    /* Initialize user application's Stack Pointer & Jump to user application */
    JumpToApplication = (pFunction)(*(__IO uint32_t *)(APPLICATION_ADDRESS + 4));

    __set_MSP(*(__IO uint32_t *)APPLICATION_ADDRESS);

    JumpToApplication();

    /* We should never get here as execution is now on user application */
    while (1)
    {

    }
}

/* BOOT  & APP Code*/

#define g_JumpInit *(uint32_t *)0x20000000

#define BOO
#ifdef BOOT
int main()
{
    if(g_JumpInit == 0x12345678)
    {
        g_JumpInit = 0;
        BSP_JumpToApp();
    }
    BSP_CPUCacheEnable();
    HAL_Init();
    BSP_SystemClockConfig();
    BSP_Usart1Init();
    while (1)
    {
        printf("BOOT START INTO APP\r\n");  
        g_JumpInit = 0x12345678;
        NVIC_SystemReset();	/* RESET CPU */
    }
}
#else
int main()  
{
    BSP_MPUConfig();
    BSP_CPUCacheEnable();
    HAL_Init();
    BSP_SystemClockConfig();
    BSP_Usart1Init();
    while (1)
    {
        printf("APP OK\r\n");
        HAL_Delay(100);
    }
}
#endif

RKOUSTM, 2020-12-15:
Hi @fyyxxm, 

Thank you for you contribution. In order to allow a better analysis of this problem, could you please give us more details about the project or the application that you are used to reproduce this issue, so that our development teams can give you an answer. Thank you again for your contribution.

With regards,

fyyxxm, 2020-12-16:
Template Project  : https://github.com/fyyxxm/BootAndApp.git

In application, if you initialize UART, HAL_Delay() will get stuck. But in BOOT, there is no problem.

In fact, HAL_Delay was the only one with a problem, and everything else was normal

RKOUSTM, 2020-12-17:
Hi @fyyxxm,

Thank you for your contribution. I cloned your [template project](https://github.com/fyyxxm/BootAndApp/tree/master/STM32H7xx_Driver) and modified it to correctly compiled and supported ARM V6, as follows :

1. Modification in the the target options is proposed :

    - ``Optimization``: -O3,
    - ``Language C``: c11,
    - ``RAM for Algorithm``: 
              - ``Size``: 0x1000

2. Update the [stm32h7xx_hal_def.h] file as follow (https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h#L110) file as follow:

```patch
@@ -110,8 +110,8 @@ 
- #if  defined ( __GNUC__ )
+ #if defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050) /* ARM Compiler V6 */
+   #ifndef __weak
+      #define __weak  __attribute__((weak))
+   #endif
+   #ifndef __packed
+      #define __packed  __attribute__((packed))
+   #endif
+ #elif defined ( __GNUC__ ) && !defined (__CC_ARM) /* GNU Compiler */
```
```patch
@@ -121,8 +121,8 @@ /* Macro to get variable aligned on 4-bytes, for __ICCARM__ the directive "#pragma data_alignment=4" must be used instead */
- #if defined   (__GNUC__)        /* GNU Compiler */
+ #if defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050) /* ARM Compiler V6 */
+   #ifndef __ALIGN_BEGIN
+     #define __ALIGN_BEGIN
+   #endif
+   #ifndef __ALIGN_END
+     #define __ALIGN_END      __attribute__ ((aligned (4)))
+   #endif
+ #elif defined ( __GNUC__ ) && !defined (__CC_ARM) /* GNU Compiler */
```
```patch
@@ -181,16 +181,16 @@ 
+/**
+ * @brief  __NOINLINE definition
+ */
+ #if defined ( __CC_ARM   ) || (defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)) || defined   (  __GNUC__  )
+ /* ARM V4/V5 and V6 & GNU Compiler
+  -------------------------------
+ */
+ #define __NOINLINE __attribute__ ( (noinline) )

+ #elif defined ( __ICCARM__ )
+ /* ICCARM Compiler
+   ---------------
+ */
+ #define __NOINLINE _Pragma("optimize = no_inline")

+ #endif
```

After this proposed modification, there is no problem when initializing the UART by calling the ``BSP_Usart1Init()`` function in ``main_app.c`` file. 

Please let us know your feedback so that we can help you. Thank you again for your contribution.

With regards,

fyyxxm, 2020-12-18:
After my test, there are still some problems with the modified project.

`HAL_Delay()` in `main_app.c`  is normal only if UART is initialized in `main_boot.c`.

If UART is not initialized in `main_boot.c`, `HAL_Delay()` will still be stuck in `main_app.c`.

It already does not affect the use for me, but it's still a problem that shouldn't exist

Thanks again for your help

minh7a6, 2020-12-19:
What I'm seeing is that you have a BSP_Disable_IRQ() in your boot code, which in your comment is to disable the Systick Interrupt. HAL relies on this interrupt to do the HAL_Delay() while LL_mDelay() use the cnt of the Systick. I think the solution is to enable the irq in your main code or not disable the irq in the boot code.

fyyxxm, 2020-12-20:
> What I'm seeing is that you have a BSP_Disable_IRQ() in your boot code, which in your comment is to disable the Systick Interrupt. HAL relies on this interrupt to do the HAL_Delay() while [...]

Thank you for your answer. I have tried this solution before, but I still haven't solved the problem.

RKOUSTM, 2020-12-21:
Hi @minh7a6,

Thank you for your contribution. Please find attached an updated version for your template project. 

[BootAndApp_Issue_102.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/5726157/BootAndApp_Issue_102.zip)

I hope this fix addresses your need. Thank you for you feedback.

With regards,

closed 2020-12-23 by fyyxxm