STMicroelectronics/STM32CubeH7 #103: Issue Entering DFU Mode after User code Sets ClockSource for SPI, Uarts, OR i2c

PGIII, 2020-12-30:
**Describe the set-up**
 Custom H750 Based Board

**Describe the bug**
H7 Wont enter DFU mode after Clock source has been set for Communication Peripherals

**How To Reproduce**
Init Clock Sources for various Peripherals
```c 
RCC_PeriphCLKInitTypeDef rccPeriphStruct;

  rccPeriphStruct.PeriphClockSelection      =
      RCC_PERIPHCLK_USB |
      RCC_PERIPHCLK_USART16 |
      RCC_PERIPHCLK_USART234578 |
      RCC_PERIPHCLK_SPI123 |
      RCC_PERIPHCLK_SPI45 |
      RCC_PERIPHCLK_QSPI |
      RCC_PERIPHCLK_ADC |
      RCC_PERIPHCLK_I2C123 |
      RCC_PERIPHCLK_I2C4 |
      RCC_PERIPHCLK_SDMMC;
  rccPeriphStruct.PLL2.PLL2M                = 5;
  rccPeriphStruct.PLL2.PLL2N                = 500;
  rccPeriphStruct.PLL2.PLL2P                = 2; 
  rccPeriphStruct.PLL2.PLL2Q                = 5; 
  rccPeriphStruct.PLL2.PLL2R                = 4;   
  rccPeriphStruct.PLL2.PLL2FRACN            = 0;
  rccPeriphStruct.PLL2.PLL2RGE              = RCC_PLL2VCIRANGE_0;
  rccPeriphStruct.PLL2.PLL2VCOSEL           = RCC_PLL2VCOWIDE;

  rccPeriphStruct.PLL3.PLL3M                = 8;
  rccPeriphStruct.PLL3.PLL3N                = 192;
  rccPeriphStruct.PLL3.PLL3P                = 1; 
  rccPeriphStruct.PLL3.PLL3Q                = 4; 
  rccPeriphStruct.PLL3.PLL3R                = 2; 
  rccPeriphStruct.PLL3.PLL3FRACN            = 0;
  rccPeriphStruct.PLL3.PLL3RGE              = RCC_PLL3VCIRANGE_0;
  rccPeriphStruct.PLL3.PLL3VCOSEL           = RCC_PLL3VCOWIDE;

  rccPeriphStruct.UsbClockSelection         = RCC_USBCLKSOURCE_HSI48;

  rccPeriphStruct.QspiClockSelection        = RCC_QSPICLKSOURCE_PLL2; 
  rccPeriphStruct.SdmmcClockSelection       = RCC_SDMMCCLKSOURCE_PLL2;

  rccPeriphStruct.Spi123ClockSelection      = RCC_SPI123CLKSOURCE_PLL;
  rccPeriphStruct.Spi45ClockSelection       = RCC_SPI45CLKSOURCE_PLL2;

  rccPeriphStruct.Usart234578ClockSelection = RCC_USART234578CLKSOURCE_PLL2;
  rccPeriphStruct.Usart16ClockSelection     = RCC_USART16CLKSOURCE_PLL2;

  rccPeriphStruct.AdcClockSelection         = RCC_ADCCLKSOURCE_CLKP;

  rccPeriphStruct.I2c123ClockSelection      = RCC_I2C123CLKSOURCE_PLL3;
  rccPeriphStruct.I2c4ClockSelection       = RCC_I2C4CLKSOURCE_PLL3;

  HAL_RCCEx_PeriphCLKConfig(&rccPeriphStruct); 
```

Set unique number in backup ram for startup code to look for
Call `HAL_RCC_DeInit` and `NVIC_SystemReset`
Startup Code checks for special code in ram and jumps to System Bootloader Location
FC is running in system bootloader but does not enumerate as USB device (Verified by pausing and checking current running location with debugger)

**Additional context**
I was able to fix this issue by setting the clock selections of any peripheral i changed back to their default value before resetting, This seems like something HAL_RCC_DeInit should do. Seems like this issue is that the sources are not set how the Bootloader expects and is getting false readings on the peripherals causing it to run a different routine than the USB one

RKOUSTM, 2021-02-09:
Hi @PGIII,

Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us more details about the application that you are using to reproduce this issue, so that our development teams can give you an answer.
Could you please confirm you are using the latest firmware version and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

RKOUSTM, 2021-03-12:
Hi @PGIII,

Please allow me to close this thread as no activity has been registered since mostly a month. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-12 by RKOUSTM