STMicroelectronics/STM32CubeH7 #104: Ethernet error interrupt

ashchepkov, 2021-01-15:
Hello!
Into stm32h7xx_hal_eth.c used macro __HAL_ETH_MAC_GET_IT:

/* ETH MAC Error IT */
  if(__HAL_ETH_MAC_GET_IT(heth, (ETH_MACIER_RXSTSIE | ETH_MACIER_TXSTSIE)))
  {
.....
}

But, macro code is 
(((__HANDLE__)->Instance->MACISR &( __INTERRUPT__)) == ( __INTERRUPT__)) .

Need two active interrupt flags simultaneously to call error handler.

With best regards.

ASELSTM, 2021-01-18:
Hi @ashchepkov,

Allow me first to thank you for your contribution. Your report has been forwarded to our development team. We will get back to you as soon as we have more details.

With regards,

ALABSTM, 2021-12-10:
Hi @ashchepkov,

We just made available a fully reworked ETH HAL driver via pull-request [stm32h7xx_hal_driver#16](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/pull/16). You can use this new driver for your applications. Please note that this new ETH HAL driver **breaks the compatibility** with the previous one.

Regarding the point you raised, it has been addressed as you will notice.  Please feel free to provide your feedback [here](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/pull/16) once you have performed your tests. 

With regards,

ALABSTM, 2022-04-05:
Hi @ashchepkov,

The new ETH HAL driver has been made officially available in the frame of **v1.10.0** of the STM32CubeH7 firmware. Please allow me to close this issue.

With regards,

closed 2022-04-05 by ALABSTM