STMicroelectronics/STM32CubeH7 #105: HAL_TIMEOUT in USB_CoreReset() due to missing volatile for count variable

adel-mamin, 2021-01-27:
Please consider making this variable `volatile`: https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1298 and in all relevant places for other micro controllers.

In my custom board I have spent quite some time figuring out why `USB_CoreReset()` failed with timeout. It turned out to be a compiler optimization. Adding `volatile` for the loop variable resolved my issue.

ALABSTM, 2021-04-20:
Hi @adel-mamin,

Thank you for reporting. You described an issue with the `USB_CoreReset()` function (the prototype of which is shown below) and mentioned a *loop variable*.
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1326

However, at line 1298 you indicate, there is a local variable `gSNPSiD` declared within function `USB_EP0_OutStart()`. Besides, this function contains no loop.
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1293

Would you please clarify how setting `gSNPSiD` volatile in `USB_EP0_OutStart()` would solve the issue you experienced with `USB_CoreReset()`? Or maybe is there simply a typo somewhere in your comment? Thank you for the clarification.

With regards,

adel-mamin, 2021-04-20:
> Would you please clarify how setting gSNPSiD volatile in USB_EP0_OutStart() would solve the issue you experienced with USB_CoreReset()? Or maybe is there simply a typo somewhere in your comment? Thank you for the clarification.

My bad. Apparently I pasted the link to a wrong place in the source code.
Here is the correct one:
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1328

Declaring `count` variable as volatile fixed my issue. As I mentioned there are other places in the source code, where it could be needed. For example:

https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L461

etc.

ALABSTM, 2021-04-26:
Hello @adel-mamin,

Would you mind precising the IDE and the compiler you are using and their versions please?

Thank you,

adel-mamin, 2021-04-26:
> Hello @adel-mamin,
> 
> Would you mind precising the IDE and the compiler you are using and their versions please?
> 
> Thank you,

\> arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ALABSTM, 2021-04-28:
Hi @adel-mamin,

We used Cube IDE **v1.6.0** with C/C++ GCC Cross Compiler Support **9.11.1**.202004012021 to build the [USB_Device/CDC_Standalone](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H743I-EVAL/Applications/USB_Device/CDC_Standalone) application with a high optimization level and **found no issue** at run-time. As hardware, we used a STM32H743I-EVAL board.

We also noticed the same use of a `count` local variable in a `do-while` loop elsewhere in the code. Actually, there are **5** functions this way implemented within the `stm32h7xx_ll_usb.c` file (listed below).

If the root-cause were the one you pointed out, you should also have had issues with all the below-listed functions. Should you not?

Would you mind checking again from your side. This reply may help you considering the issue from another point-of-view.

With regards,

https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L437
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L459
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1326
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1864
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c#L1961

adel-mamin, 2021-04-28:
> If the root-cause were the one you pointed out, you should also have had issues with all the below-listed functions. Should you not?

Yes, in my project I have added `volatile` to all five places you pointed out.
FYI, I've been using a custom board based on STM32H747IIT6 SoC. 

Also here are the relevant command line parameters of my gcc invocation for that file:

```
/usr/sbin/arm-none-eabi-gcc -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O3 -g -Wpedantic -pedantic-errors -Wall -Wextra -Wchar-subscripts -Wformat -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-y2k -Winit-self -Wmain -Wmissing-include-dirs -Wswitch-default -Wswitch-enum -Wunused -Wunused-parameter -Wunused-variable -Wuninitialized -Wunknown-pragmas -Wstrict-overflow=5 -Wmissing-format-attribute -Wfloat-equal -Wshadow -Wpointer-arith -Wundef -Wunused-macros -Wcast-qual -Wcast-align -Wwrite-strings -Wconversion -Wdangling-else -Wdate-time -Wmissing-declarations -Wpacked -Wredundant-decls -Winline -Wdisabled-optimization -Werror -Winvalid-pch -Wmissing-noreturn -Wunreachable-code -Wimplicit -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -Wnested-externs -Wnull-dereference -fno-strict-aliasing -fno-common -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -fno-asynchronous-unwind-tables -ftrapv -mthumb -mabi=aapcs -Wdouble-promotion -mcpu=cortex-m7 -mno-unaligned-access -mfloat-abi=hard -ggdb -mfpu=fpv5-sp-d16 -DUSE_HAL_DRIVER -DSTM32H747xx -DUSE_FULL_LL_DRIVER -DCORE_CM7 -MD -MQ ... STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c
```

radioflash, 2021-07-14:
Making the count variable volatile should not be necessary, because the loop condition itself is volatile, so the compiler should NEVER be allowed to return timeout after a single check.

@adel-mamin Maybe the __IO macro (expected to be `#define __IO volatile`) is empty/broken for some reason in your build? This macro would normally qualify the loop variable as volatile, defective definitions for it would explain why you need the volatile `count` instead.

atsju, 2021-08-18:
I confirm what @radioflash explaines. The outside loop condition should be volatile and this removes the need for a volatile `count` variable.

There might be a problem and it might be "solved" with volatile but it's not the root cause.

ALABSTM, 2022-01-12:
Hi @adel-mamin,

I hope you are doing well. We were not able to reproduce the problem you described. Please allow me to close this issue. Thank you for your comprehension.

@radioflash and @atsju thank you very much for your explanation and proposal.

With regards,

closed 2022-01-12 by ALABSTM