STMicroelectronics/STM32CubeH7 #106: Export Ioc

DrunkenMastah, 2021-02-01:
Currently it is a bit difficult to modify the examples as there is no ioc file and one can not open them up on CubeMx to explore , these examples would be much more useful if you exported an ioc file with them.
That is basically a feature request and not exactly an issue , do you have any thoughts about that ?

ALABSTM, 2021-02-01:
Related to [STM32CubeL4#21](https://github.com/STMicroelectronics/STM32CubeL4/issues/21).

ALABSTM, 2021-02-01:
Hi @DrunkenMastah,

 IOC files are published within [this](https://github.com/STMicroelectronics/STM32_open_pin_data) repository. Hopefully you will find the one(s) that would address your needs under the **./boards** directory.

With regards,

ALABSTM, 2021-02-08:
Hi @DrunkenMastah,

Any feedback after my last comment? Could you find what you are looking for?

Thank you,

DrunkenMastah, 2021-02-09:
Yes thank you very much ,I still believe though that a lot of useful examples (e.g. for the Discovery board) are pretty much locked for everyone using a simple gcc toolchain.

closed 2021-02-09 by DrunkenMastah