STMicroelectronics/STM32CubeH7 #107: Incorrect comments on SDMMC_GetCmdResp7

C47D, 2021-02-02:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * Nucleo-H753ZI
 * STM32CubeIDE
 * H7 HAL 1.8.0

**Additional context**
This is a question on the comments on the `SDMMC_GetCmdResp7` function. The first two `/* Card is SD V2.0 compliant */` should be `/* Card is not SD V2.0 compliant */` instead? When `SDMMC_FLAG_CTIMEOUT` and `SDMMC_FLAG_CCRCFAIL` flags are set.

This is the current version of it.
```c
static uint32_t SDMMC_GetCmdResp7(SDMMC_TypeDef *SDMMCx)
{
  uint32_t sta_reg;
  /* 8 is the number of required instructions cycles for the below loop statement.
  The SDMMC_CMDTIMEOUT is expressed in ms */
  uint32_t count = SDMMC_CMDTIMEOUT * (SystemCoreClock / 8U /1000U);

  do
  {
    if (count-- == 0U)
    {
      return SDMMC_ERROR_TIMEOUT;
    }
    sta_reg = SDMMCx->STA;
  }while(((sta_reg & (SDMMC_FLAG_CCRCFAIL | SDMMC_FLAG_CMDREND | SDMMC_FLAG_CTIMEOUT)) == 0U) ||
         ((sta_reg & SDMMC_FLAG_CMDACT) != 0U ));

  if(__SDMMC_GET_FLAG(SDMMCx, SDMMC_FLAG_CTIMEOUT))
  {
    /* Card is SD V2.0 compliant */
    __SDMMC_CLEAR_FLAG(SDMMCx, SDMMC_FLAG_CTIMEOUT);

    return SDMMC_ERROR_CMD_RSP_TIMEOUT;
  }

  else if(__SDMMC_GET_FLAG(SDMMCx, SDMMC_FLAG_CCRCFAIL))
  {
    /* Card is SD V2.0 compliant */
    __SDMMC_CLEAR_FLAG(SDMMCx, SDMMC_FLAG_CCRCFAIL);

    return SDMMC_ERROR_CMD_CRC_FAIL;
  }
  else
  {
    /* Nothing to do */
  }

  if(__SDMMC_GET_FLAG(SDMMCx, SDMMC_FLAG_CMDREND))
  {
    /* Card is SD V2.0 compliant */
    __SDMMC_CLEAR_FLAG(SDMMCx, SDMMC_FLAG_CMDREND);
  }

  return SDMMC_ERROR_NONE;

}
```

**Screenshots**

ASELSTM, 2021-02-04:
Hi @C47D,

Thank you for this report. It has been forwarded to our development teams. I will get to you as soon as they provide their feedback.

With regards,

ASELSTM, 2021-02-26:
Hi @C47D,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-02-26:
ST Internal Reference: 102186

ALABSTM, 2021-12-16:
Hi @C47D,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub as you can see below. Thank you again for having reported.
 
https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_sdmmc.c#L1570-L1571

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_sdmmc.c#L1578-L1579

With regards,

closed 2021-12-16 by ALABSTM