STMicroelectronics/STM32CubeH7 #108: Git Repository should be split up into submodules

WBABD, 2021-02-03:
**Describe the set-up**
I am experiencing long clone times due to all dependencies being copied in directly instead of using submodules

**Describe the bug**
lwIP, FreeRTOS and CMSIS are copied in directly instead of referencing other git repositories. The examples should be contained in another module instead of with all the drivers.

**How To Reproduce**
Clone the project using Git

CCASTM, 2021-02-08:
Hello WBAD

The first intent of the GitHub publication (in April 2019) was to propose an alternative to the zip format proposed in st.com servers.
You can find then all FW Packages series under configuration management. This is a first added value to manage the successive versions from the customer point of view, an benefit from the Issues and Pull Request features.
Since December 2019, you can find also the first components published individually (CMSIS Core and Device, HAL Drivers, a first set of MW, X-CUBE ). 
Please take a look to the repo  https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer   to get a description of all STM32 git repos
Regards
Christophe

CCASTM, 2021-02-08:
Just to complete my previous comments.
The sub-modules feature was not selected to publish the FW Packages as the full packages exist as is in our internal configuration management.
In the future, we have in mind to use git sub-repo to publish specific application projects
Regards
Christophe

ALABSTM, 2022-04-05:
Hi @WBABD,

As explained by @CCASTM, not using submodules for Cube repos is a choice. On the other hand, choice has also been made to use them for other repos (a handful so far):
* [stm32u5-classic-coremw-apps](https://github.com/STMicroelectronics/stm32u5-classic-coremw-apps)
* [stm32l5-openbl-apps](https://github.com/STMicroelectronics/stm32l5-openbl-apps)
* [stm32wl-openbl-apps](https://github.com/STMicroelectronics/stm32wl-openbl-apps)

Thank you again for having tackled the subject. Please allow me to close this thread.

With regards,

closed 2022-04-05 by ALABSTM