STMicroelectronics/STM32CubeH7 #111: Impact of using TEMPSENSOR_CAL_VREFANALOG with macro __LL_ADC_CALC_TEMPERATURE  on macro __HAL_ADC_CALC_TEMPERATURE and the output Temperature

rxa1031, 2021-02-10:
Hello @RKOUSTM ,

I analyzed your response to #72 and observed that your response is correct only when user uses the below code:
```__HAL_ADC_CALC_TEMPERATURE(TEMPSENSOR_CAL_VREFANALOG, g_u32ReadTemperatureDigitalValue, ADC_RESOLUTION_16B)```

However, the response is incorrect if `TEMPSENSOR_CAL_VREFANALOG` is replaced with actual VREF value read via the VREF analog input channel value read using the ADC.

On my hardware the read VREF value is 3275 which corresponds to 3.275V (and not 3300 i.e. TEMPSENSOR_CAL_VREFANALOG which would correspond to 3.3V)

I request you to review the macro `__LL_ADC_CALC_TEMPERATURE` declaration. Is it possible for you to remove the dependency of the macro on `TEMPSENSOR_CAL_VREFANALOG` and `__VREFANALOG_VOLTAGE__`?

Thanks,
Rajeev

rxa1031, 2021-02-10:
I am comparing result / output of `__HAL_ADC_CALC_TEMPERATURE` with result of `_HAL_ADC_CALC_TEMPERATURE_TYP_PARAMS(2000, 620, 30, u16VRefMilliVoltsValue, g_u32ReadTemperatureDigitalValue, ADC_RESOLUTION_16B));`

Where:
`u16VRefMilliVoltsValue = 3275`
`g_u32ReadTemperatureDigitalValue = 13068`
`TS_CAL2 = 16234`
`TS_CAL1 = 12323`

RKOUSTM, 2021-02-10:
Hi @rxa1031,

Thank you for this contribution. Please let me introduce @ASELSTM, which has already reviewed your change to #72.

With regards,

ASELSTM, 2021-02-19:
Hi @rxa1031,

Glad to read from you again.

To answer your query, it is unfortunately impossible to remove the dependency of the macro on `TEMPSENSOR_CAL_VREFANALOG` which corresponds to the  Analog voltage reference (Vref+) voltage with which temperature sensor has been calibrated in production and `__VREFANALOG_VOLTAGE__`.  

Let me explain it in more detail,  as mentioned in the issue #72  the formula used to compute the temperature value within the `__LL_ADC_CALC_TEMPERATURE` is as following : 

[image]

My focus will be more on this part of the formula : 

[image]

In the macro, the `TS_DATA` is obtained as following : 

```
((__LL_ADC_CONVERT_DATA_RESOLUTION((__TEMPSENSOR_ADC_DATA__),     \
                                                    (__ADC_RESOLUTION__),          \
                                                    LL_ADC_RESOLUTION_16B)         \
                   * (__VREFANALOG_VOLTAGE__))                                     \
                  / TEMPSENSOR_CAL_VREFANALOG) 
```

The ratio between  `__VREFANALOG_VOLTAGE__` and `TEMPSENSOR_CAL_VREFANALOG` is essential to align the ADC conversion scale. 
The Analog reference voltage (Vref+) when calibrating the temperature sensor is 3.3V for STM32H7 case which is different from the the Analog reference voltage used in the application, **3.275V** in your case. 
The ADC conversion scales have a discrepancy of 0.8%, they have then to be aligned before comparing the ADC data with TS_CAL1.

Hopefully this would help you.

With regards

ASELSTM, 2021-03-12:
Hi @rxa1031,

Please allow me to close this thread as no activity has been registered since mostly a month. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-12 by ASELSTM