STMicroelectronics/STM32CubeH7 #112: core_cm7.h:  loses information [clang-diagnostic-error]

lantiati, 2021-02-10:
Hi,
We are running clang-tidy check our code, and the following errors are generated: 

\STM32CubeH7\Drivers\CMSIS\Include\core_cm7.h:2468:24: error: cast from pointer to smaller type 'uint32_t' (aka 'unsigned int') loses information [clang-diagnostic-error]
    uint32_t op_addr = (uint32_t)addr;
                       ^
\STM32CubeH7\Drivers\CMSIS\Include\core_cm7.h:2495:24: error: cast from pointer to smaller type 'uint32_t' (aka 'unsigned int') loses information [clang-diagnostic-error]
    uint32_t op_addr = (uint32_t) addr;
                       ^
\STM32CubeH7\Drivers\CMSIS\Include\core_cm7.h:2522:24: error: cast from pointer to smaller type 'uint32_t' (aka 'unsigned int') loses information [clang-diagnostic-error]
    uint32_t op_addr = (uint32_t) addr;

This error aborts the clang-tidy check.  

If I modify the code like below, the compilation is ok:
 **uint32_t op_addr = * addr;**

Thank you for the help!
Attila Lantos

RKOUSTM, 2021-02-10:
Hi @lantiati,

Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us more details about the application that you are using to reproduce this issue, so that our development teams can give you an answer.
Could you please confirm you are using the latest firmware version and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

lantiati, 2021-02-10:
We are using cmake, and it checks out the master branch from this repo. So the version is the latest one.

RKOUSTM, 2021-02-10:
Hi @lantiati,

Thank you for this report. Unfortunately we don't treat aspect related to ``core_cm7`` in our GitHub repositories. As the [CMSIS Core](https://github.com/STMicroelectronics/cmsis_core/tree/master/Core/Include) files. These files come from Arm and we use them as is. So, these files are out of the scope of STMicroelectronics what about reporting this issue to the [ARM](https://github.com/ARM-software) GitHub?

Please allow me to close this issue now. Thank you again for your contribution.

With regards,

closed 2021-02-10 by RKOUSTM