STMicroelectronics/STM32CubeH7 #114: TIM12 and 15 missing in several TIMER_x macro's.

basvloon, 2021-02-14:
Hello, 
after debugging my timer interface I have found the following macros are missing a check for TIM12:
-  IS_TIM_MASTER_INSTANCE(), TIM12 allows master mode setting (MMS) although it only allows to drive a DMA 'channel'.
- IS_TIM_CLOCKSOURCE_ITRX_INSTANCE(), TIM12 allows setting an internal input trigger.
- IS_TIM_CLOCKSOURCE_TIX_INSTANCE(), TIM12 supports external clock mode 1.
- 
Maybe there are more issues regarding TIM12 but did not check further. I can also confirm issue reported at [https://github.com/STMicroelectronics/STM32CubeH7/issues/91](url).

Potential patch to fix TIM12 checks:
`
--- STM32H7xxxx.orig/CMSIS_HAL/Device/ST/STM32H7xx/Include/stm32h750xx.h        2021-02-14 12:48:21.997908200 +0800
+++ STM32H7xxxx/CMSIS_HAL/Device/ST/STM32H7xx/Include/stm32h750xx.h     2021-02-14 13:08:38.862692100 +0800
@@ -26124,6 +26124,7 @@
                                             ((INSTANCE) == TIM6)   || \
                                             ((INSTANCE) == TIM7)   || \
                                             ((INSTANCE) == TIM8)   || \
+                                            ((INSTANCE) == TIM12)  || \
                                             ((INSTANCE) == TIM15))

 /****** TIM Instances : Salve mode available (TIMx_SMCR.TS available )*********/
@@ -26300,6 +26301,7 @@
    ((INSTANCE) == TIM4)    || \
    ((INSTANCE) == TIM5)    || \
    ((INSTANCE) == TIM8)    || \
+   ((INSTANCE) == TIM12)   || \
    ((INSTANCE) == TIM15)   || \
    ((INSTANCE) == TIM16)   || \
    ((INSTANCE) == TIM17))
@@ -26330,6 +26332,7 @@
    ((INSTANCE) == TIM4)    || \
    ((INSTANCE) == TIM5)    || \
    ((INSTANCE) == TIM8)    || \
+   ((INSTANCE) == TIM12)   || \
    ((INSTANCE) == TIM15))

 /****************** TIM Instances : supporting internal trigger inputs(ITRX) *******/
@@ -26340,6 +26343,7 @@
    ((INSTANCE) == TIM4)    || \
    ((INSTANCE) == TIM5)    || \
    ((INSTANCE) == TIM8)    || \
+   ((INSTANCE) == TIM12)   || \
    ((INSTANCE) == TIM15))

 /****************** TIM Instances : supporting OCxREF clear *******************/
`

Hope this helps.

Best Regards,
Bas

ASELSTM, 2021-02-15:
Hi @basvloon,

First we would like to thank you for pointing this out.

Regarding the first macro `IS_TIM_MASTER_INSTANCE()`,  the issue has already been dealt and fixed internally. The fix will be part of the next release.

For the remaining macros, `IS_TIM_CLOCKSOURCE_ITRX_INSTANCE()` and `IS_TIM_CLOCKSOURCE_TIX_INSTANCE()`, you're absolutely right, TIM12 is missing and should be added.  This fix will be made available in the frame of a next release as well as the fix of the issue #91 which has been also confirmed . 

Thank you once again for your contribution and hope to hear from you again.

With regards,

ASELSTM, 2021-02-15:
ST Internal Reference: 101417

ASELSTM, 2021-02-15:
ST Internal Reference: 98873

ALABSTM, 2021-12-16:
Related to #91.

ALABSTM, 2021-12-16:
Hi @basvloon,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-16 by ALABSTM