STMicroelectronics/STM32CubeH7 #115: SPI and DMA is broken (no completion callback and stuck BUSY)

tringis, 2021-02-22:
HAL_SPI_Transmit_DMA is broken for H7 because the callback SPI_DMATransmitCplt() in stm32h7xx_hal_spi.c doesn't really do anything unless Mode is DMA_CIRCULAR.

Specifically, it doesn't call the completion callback HAL_SPI_TxCpltCallback() and it also doesn't set the State to HAL_SPI_STATE_READY (which causes the next call to HAL_SPI_Transmit_DMA to fail).

The functions HAL_SPI_Receive_DMA() and HAL_SPI_TransmitReceive_DMA() have the same problem.

**How To Reproduce**

1. Compile test program "SPI_FullDuplex_ComDMA" for NUCLEO-H723.
2. Set a breakpoint on HAL_SPI_TxRxCpltCallback
3. Run the program
4. The breakpoint is never reached

ASELSTM, 2021-02-26:
Hi @tringis,

First we would like to thank you for your contribution.

We have checked the example **SPI_FullDuplex_ComDMA** for NUCLEO-H723 and it is working correctly. The example did access the `HAL_SPI_TxRxCpltCallback` where LED1 and LED2 are activated to notify that transfer is completed.  

Please make sure you are correctly connecting the Master and Slave boards as mentioned in the [readme](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H723ZG/Examples/SPI/SPI_FullDuplex_ComDMA/readme.txt).

With regards,

ASELSTM, 2021-03-17:
Hi @tringis,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-17 by ASELSTM