STMicroelectronics/STM32CubeH7 #118: UART Receive with timeout interrupt issue

rxa1031, 2021-03-09:
Dear @RKOUSTM 

I checked your response for #68 and #31 

Used STM32CubeMX version 6.2.0 with STM32CubeMX along with latest released STM32CubeH7 package version **1.9.0**

**I still observe an issue for a particular scenario.**

**Good Behavior / Scenario**
At least one byte must have been received via the Asynchronous UART port after a call to `HAL_UART_Receive_IT() API` is made.
In this scenario the _Receiver Timeout interrupt triggers_ after the specified time,

**BAD Behavior / Scenario**
If _NOT EVEN A SINGLE BYTE_ is received over Asynchronous UART after the `HAL_UART_Receive_IT() API` is executed, then the `Receiver Timeout (RTOF) interrupt` present with the UART related `USART_ISR` **NEVER SETS** and hence the _interrupt never triggers_ even though the related bits in USART_CR1 and USART_CR2 are set.

Please help me understand how to fix above issue.

With help of Receiver Timeout interrupt I want to ensure that:
1. The connected device started to communicate within a defined time period.
2. The response was not sent by connected device and hence data need to be retransmitted
3. Possibly loose connection or cable break

Please suggest what changes need to be made.

Thanks,
Rajeev

rxa1031, 2021-03-09:
Sample code attached for reference:
[Gateway-H7-App-with-STM32CubeH7-ver-1.9.0-from-scratch-20210309-2311-IST.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6110532/Gateway-H7-App-with-STM32CubeH7-ver-1.9.0-from-scratch-20210309-2311-IST.zip)

ASELSTM, 2021-03-11:
Hi @rxa1031,

The Receiver Timeout event is triggered when line has been in idle state for a programmable time. Receiver timeout is expressed in terms of bit duration and it is applicable only once the transfer occurred. 

In fact, the receiver timeout counter starts counting from the received stop bit, in other words, as soon as a character is received.
As described in the [reference manual ](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) in the **"USART receiver timeout register (USART_RTOR)"**  section :

> In standard mode, the RTOF flag is set if, after the last received character, no new start bit is detected for more than the RTO value.

This is explained in more details within the  **"USART receiver timeout"** section :

> The receiver timeout counter starts counting:
> • from the end of the stop bit if STOP = ‘00’ or STOP = ‘11’
> • from the end of the second stop bit if STOP = ‘10’.
> • from the beginning of the stop bit if STOP = ‘01

So that the Receiver Timeout feature can't be used if no character have been received.  The interruption will never be trigged even though  `USART_CR1_RTOIE` and `USART_CR2_RTOEN` bits are set and an appropriate receiver timeout value is configured in the `USART_RTOR` register.

Hope this would help you. 

With regards,

ASELSTM, 2021-03-17:
Hi @rxa1031,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-17 by ASELSTM