STMicroelectronics/STM32CubeH7 #121: Ethernet: write to SYSCFG before MAC reset require wait

amurzeau, 2021-03-12:
**Describe the set-up**
 * Board: NUCLEO-H743ZI
 * IDE: STM32CubeIDE v1.5.1
 * STM32CubeH7 Firmware v1.8.0

**Describe the bug**
When using a system clock of 400 Mhz, a HPRE prescaler of /2 and a APB4 divisor of /16, the ethernet MAC does not initialize correctly. In that case, ethernet does not work (no ping from a PC)

**How To Reproduce**
1. Generate a project using LwIP
2. Configure a /16 APB4 prescaler and a /2 HPRE prescaler
3. Run the project and try to ping the STM32 from a PC
4. The ping fails
5. Change the APB4 prescaler to something like /2 and reflash the board
6. The ping works

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)

In [stm32h7xx_hal_eth.c in function HAL_ETH_Init](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_eth.c#L327), when adding `(void)SYSCFG->PMCR` like this, the ethernet MAC is correctly initialized.
```c
  if(heth->Init.MediaInterface == HAL_ETH_MII_MODE)
  {
    HAL_SYSCFG_ETHInterfaceSelect(SYSCFG_ETH_MII);
  }
  else
  {
    HAL_SYSCFG_ETHInterfaceSelect(SYSCFG_ETH_RMII);
  }

  (void)SYSCFG->PMCR; // Adding this line will fix the issue and make ethernet works again

  /* Ethernet Software reset */
  /* Set the SWR bit: resets all MAC subsystem internal registers and logic */
  /* After reset all the registers holds their respective reset values */
  SET_BIT(heth->Instance->DMAMR, ETH_DMAMR_SWR);
```

The reason is probably that:
 * SYSCFG peripheral uses the APB4 peripheral clock
 * When setting the RMII/MII configuration, it takes more clock pulses to be taken into account because of the /16 APB4 prescaler (and maybe also /2 HPRE prescaler)
 * When executing the Ethernet Software reset, the SYSCFG RMII/MII muxes might not be configured yet
 * As state in the datasheet for the DMAMR register, bit SWR, doing the ethernet software reset requires to have all PHY input clocks established:
> The reset operation is complete only when all resets in all active clock domains are deasserted. Therefore, it is essential that all PHY inputs clocks (applicable for the selected PHY interface) are present for software reset completion. The time to complete the software reset operation depends on the frequency of the slowest active clock.

So to ensure all PHY inputs clocks are present, the RX and TX clocks must be correctly muxed via the SYSCFG block.
This requires to add a dummy read or a delay to ensure the write to SYSCFG->PMCR[EPIS] is taken into account before doing the ethernet software reset.

RKOUSTM, 2021-03-15:
Hi @amurzeau,

Thank you for your contribution. Could you confirm that you are using the latest firmware version `(v1.9.0)` recently published on GitHub and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

amurzeau, 2021-03-17:
Hi,

Yes but the issue is the same and it is systematically reproducible.
`(void)SYSCFG->PMCR;` dummy read is required to fix it (or anything that will wait the previous SYSCFG write to be completed)

RKOUSTM, 2021-08-06:
Hi @amurzeau,

I hope you are fine. Sorry for the delay it may take sometimes to answer. A new version of the ETH HAL driver will take this problem into into account. 

Thank you again for this contribution and thank you in advance for your patience.

With regards,

closed 2022-07-08 by ASELSTM