STMicroelectronics/STM32CubeH7 #122: Concern: Asynchronous USART1 TX data mismatch when using Parity

rxa1031, 2021-03-15:
Hello,

I am using the latest released STM32CubeH7 version (i.e. 1.9.0)

I was trying to use the STM32CubeMX sample code of wakeup from sleep with help of USART1 in asynchronous mode for  STM32H743I-Eval board with version Y of Microcontroller.

But I did not observe any communication on Serial, even after connecting board's CN2 to Prolific Serial to USB converter.

Hence, I created the attached project
[UART_wake_from_Sleep.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6141208/UART_wake_from_Sleep.zip)

This project hex file when flashed to microcontroller should sent out text stated within Quotes "Q12w". However the character 1 gets modified.

Used settings 9600 bps 7 bit data, Odd parity, 1 stop bit.

Attached below is the HyperTerminal view:

[image]

 When I changed the display text to HEX values (for ease of comparison) and also changed HyperTerminal view to show HEX values, I observe that instead of **0x31** (ie. 1) **0x19** (i.e. End Of Medium) is being printed on _HyperTerminal_.

Please suggest a Fix.

Thanks,
Rajeev

rxa1031, 2021-03-15:
Two more observations:

1. When using Debugger (ST-Link/V2-1), the character NULL does not precede the string. However, when microcontroller is reset (and Debugger not used), the NULL character always precedes the string.
2. I even used Tera Term and found similar result: 
[image]

For testing only I changed Number of Data bits to 8 (on HyperTerminal) and the output was different.

Regards,
Rajeev

rxa1031, 2021-03-15:
Please change string in code to below, the character 'x' accidently got deleted before zipping the code.

HAL_UART_Transmit_IT(&huart1, (uint8_t *)"\x51\x31\x32\x77", 4);

ASELSTM, 2021-03-16:
Hi @rxa1031,

Glad to read from you again.

Unfortunately, we couldn't reproduce the issue you have described.  We have tested the example you have shared and added the missing 'x' as you have mentioned in the [last comment](https://github.com/STMicroelectronics/STM32CubeH7/issues/122#issuecomment-799416373). The message is well printed on the Hyper Terminal as you may see in the screenshot below. 

[image]

Actually, the missing 'x' is the root of the mismatching message you got within the Hyper Terminal. The character was interpreted in an erroneous way, hence the output you have obtained.

With regards,

ASELSTM, 2021-03-25:
Hi @rxa1031,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-25 by ASELSTM