STMicroelectronics/STM32CubeH7 #123: HAL_UART_Receive_IT either captures more data then requested or does not capture data in certain conditions.

rxa1031, 2021-03-18:
Hello,

I am using STM32CubeH7 release version 1.9.0

I am observing that HAL_UART_Receive_IT either captures more data then requested or does not capture data in certain conditions.

Please refer to the attached XLSX document for details and snapshots
[HAL_UART_Receive_IT-Issues.xlsx](https://github.com/STMicroelectronics/STM32CubeH7/files/6163864/HAL_UART_Receive_IT-Issues.xlsx)

I am also attaching the test code which uses the macros described in the XLSX document. At any given time one macro for FIFO can be chosen and one macro for CTS and RTS state can be chosen.
[Gateway-H7-App-with-STM32CubeH7-ver-1.9.0-from-scratch.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6163875/Gateway-H7-App-with-STM32CubeH7-ver-1.9.0-from-scratch.zip)

Please help me with a fix for the issue.

Thanks,
Rajeev

rxa1031, 2021-03-18:
I slowed down the baud rate to 300 bps and chose a large sized Tx data to test this scenario. this helped to simulate scenarios wherein STM32H7xx might me transmitting a message to external device and the external device may respond with outputs like answer, ACK, NACK, etc.
There could be scenarios wherein the external device transmitted message before the STM32H7xx was even able to initiate a UART read. also there can be cases where the UART read has already been initiated.

Hence I have tried to test all scenarios and behavior of the HAL_UART_Receive_IT API for all thinkable conditions.

Regards,
Rajeev

ASELSTM, 2021-03-23:
Hi @rxa1031,

Thank you for your contribution and for this detailed report. We would also like to thank you for the effort and time you put into testing and synthesizing all these scenarios. We do really appreciate that our GitHub community members actively contribute to the enhancement of our software. 

This issue is going to be thoroughly investigated.  We will get back to you as soon as possible.

With regards,

rxa1031, 2021-05-07:
Dear @ASELSTM 

Please share any available update regarding this concern.

Thanks,
Rajeev

rxa1031, 2021-09-13:
Dear @ALABSTM 

Will the fix for this issue be available with STM32CubeH7 release version 1.10.0?

Regards,
Rajeev

ASELSTM, 2023-04-06:
Hi @rxa1031,

Would you please try to check with the latest version of the STM32CubeH7.
Please allow me then to close this thread. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-04-06 by ASELSTM