STMicroelectronics/STM32CubeH7 #126: Query: With Keil project, unable to us "No Auto Include" option

rxa1031, 2021-03-22:
Hello,

I am using STM32CubeH7 version 1.9.0. I am using Keil Version 6 compiler. I have manually removed the RVDS folder and its sub folder. I have added GCC and appropriate subfolders that have the port.c file.

I wanted to ensure that anyone who modifies the user code must define the path to the include files that is to be included, as there can be scenarios wherein a known macro might have different value assigned to it in different files, because of the limited scope of the header file.

I therefore set the "_No Auto Include_" option under _Options for Target -> C/C++(AC6)_ and removed all the string mentioned next to include Paths.
[image]

If I would have not observed errors, I would have done the same for Asm Includes as well.

**Is it possible that the STM32CubeH7 C files and the assembly files use the relative path for the header file / Include file, instead of using just the header file name?**

Please suggest.

Thanks,
Rajeev

RKOUSTM, 2021-03-24:
Hi @rxa1031,

Thank you for your contribution. For the first point, in order to support ARM V6, the project settings should be updated as follows:

- [ ] C/C++(AC6) :
     - Warnings => select "AC5-like Warnings"
     -  Language C => keep the current selected version ("c99" or "c11")
     -  Language C++ => keep the current selected version ("c++11")
     - Short enum/wchar => selected  (keep compatibility to RDVS with wchar = 2bytes)
     - Optimization => select "Oz image size"
     - Misc controls => remove "--C99" if present

For the second point, to solve your problem in the` Include Paths` section 

[image]
just include the following path : 

[image]

Please let us know your feedback so that we can help you. 

With regards,

rxa1031, 2021-03-25:
Dear @RKOUSTM 

Above changes did not help.

The path are bit different in the code generated by STM32CubeMX, and removing a few of ../ may help. I am working on this as of now.

My concern however remains that as per company Guidelines, I always have to set the option **No Auto Includes** under C/C++(AC6). Thus my concern can only be resolved if each STM32CubeH7 provided file (including Drivers and FreeRTOS)uses a relative path for the header files in the STM32CubeH7 provided C files.

I can make tis change once myself. But anytime anyone from my team regenerates the code using STM32CubeMX the changes will vanish.

Thanks,
Rajeev

RKOUSTM, 2021-03-25:
Hi @rxa1031,

Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us the .ioc file that you are used to reproduce this issue. Thank you again for your contribution.

With regards,

rxa1031, 2021-03-25:
Dear @RKOUSTM ,

I have created a very simple project with help of STM32CubeMX latest version. I am sharing the IOC file and the project for reference:
[Keil-AC6-No-Auto-Includes.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6203230/Keil-AC6-No-Auto-Includes.zip)

For reference, I have left the file corresponding to Build Output in folder MDK-ARM\Keil-AC6-No-Auto-Includes.

As the latest STM32CubeMX is not capable of generating a project that can directly be compiled using the _Keil Version 6 Compiler_, hence I manually change project settings. I set the "Code Generator ARM Compiler" / compiler settings to _Use default compiler version 6_ (under Targets). I then make the necessary C/C++(AC6) settings (including No Auto Includes)

Then I delete the _port.c_ file link from group located at Project Pane -> Project: <Project Name> -> <Project Name> -> Middlewares/FreeRTOS

Then I take the latest GCC\ARM_CM4F folder along with its contents and past it at the same location where the RVDS folder is present.

The new path to _port.c_ will look something like <folder path to project>\Keil-AC6-No-Auto-Includes\Middlewares\Third_Party\FreeRTOS\Source\portable\GCC\ARM_CM4F\port.c

I include the above informed file to the Middlewares/FreeRTOS Group and then finally COMPILE the code.

Regards,
Rajeev

RKOUSTM, 2021-03-25:
Hi @rxa1031,

Thank you for your contribution. The compilation result of your template project is as follows :

[image]

To solve your reported problem, an update on the the `Include Paths` section is proposed : 
[image]

Please let us know your feedback so that we can help you. Thank you again for your contribution.

With regards,

RKOUSTM, 2021-03-25:
Hi @rxa1031,

Thank you for your contribution. After investigation, the point you reported is related to the `CubeMX` generation problem. In fact, when you regenerate the project after updating the path as follows : 

[image]

CubeMX will always add the old path which generates a compilation error. 

[image]

This point has ben reported to `CubeMX development teams` who will take in charge the issue. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release of 'CubeMX'. 

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-03-25:
ST Internal Reference: 103818

rxa1031, 2021-03-27:
Hello,

I have another observation to share. The STM32CubeMX does not check that path for port.c file was changed to use file from GCC/ARM_CM4F. It additionally adds path for RVDS/ARM_CM4F/port.c to the project. This then causes issues during compilation.

The tools also adds the RVDS/ARM_CM4F folder and is content back to the project folder even though the GCC folder and its subfolder along with files exists.

Thanks,
Rajeev

rxa1031, 2021-03-27:
Please have a look at the below discussion:
https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/154

ALABSTM, 2021-12-16:
Hi @rxa1031,

I hope you are fine. The issue you reported has been logged into an internal bug tracker. However, as it is related to the ecosystem rather than the firmware published within this repository, please allow me to close it.

In case you have other points to share, please submit them in separate issues, one per point ideally. Thank you for your comprehension and thank you again for having reported.

With regards,

closed 2021-12-16 by ALABSTM