STMicroelectronics/STM32CubeH7 #127: USART_CR2.RTOEN check missing

rxa1031, 2021-03-25:
Hello,

In the datasheet I observe below information when checking bits available with **USART_CR2**:
_Bit 23 **RTOEN**: Receiver timeout enable
This bit is set and cleared by software.
0: Receiver timeout feature disabled.
1: Receiver timeout feature enabled.
When this feature is enabled, the RTOF flag in the USART_ISR register is set if the RX line is idle
(no reception) for the duration programmed in the RTOR (receiver timeout register)._

I however do not observe the check added to below code in the ISR service routine:
https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L2294

Please suggest whether the code must be modified.

Thanks,
Rajeev

rxa1031, 2021-03-25:
Assuming I am using the LL Driver, should the updated check be:

```
	if( LL_USART_IsEnabledRxTimeout(phLocalUART) && LL_USART_IsEnabledIT_RTO(phLocalUART) && LL_USART_IsActiveFlag_ORE(phLocalUART) )
	{
		LL_USART_ClearFlag_ORE(phLocalUART);
	}

```

ASELSTM, 2021-03-25:
Hi @rxa1031,

Referring to the HAL driver, the **USART_CR1_RTOIE** bit could be enabled within the [HAL_UART_Receive_DMA()](https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L1489) and [HAL_UART_Receive_IT()](https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L1370) APIs after checking on the **USART_CR2_RTOEN** state as follows : 

https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L1506-L1512

The USART_CR1_RTOIE state is closely dependent on the state of USART_CR2_RTOEN bit. Thus, receiver timeout interrupt can only be enabled if **USART_CR2_RTOEN** is already set. Hence, if the receiver timeout feature is enabled. 

Therefore, there is  no need to add a check on USART_CR2_RTOEN state in the ISR service routine since if USART_CR1_RTOIE is active then so is USART_CR2_RTOEN bit. 

With regards,

ASELSTM, 2021-04-06:
Hi @rxa1031,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-04-06 by ASELSTM