STMicroelectronics/STM32CubeH7 #128: Wrong description for timer one pulse mode

basvloon, 2021-03-27:
Hello,

browsing through the timer header I spotted that the description of the one pulse mode is swapped for single and repetive mode.

/** @defgroup TIM_LL_EC_ONEPULSEMODE One Pulse Mode
  * @{
  */
#define LL_TIM_ONEPULSEMODE_SINGLE             TIM_CR1_OPM          /*!< Counter is not stopped at update event */
#define LL_TIM_ONEPULSEMODE_REPETITIVE         0x00000000U          /*!< Counter stops counting at the next update event */
/**
  * @}
  */

I think this should be:
/** @defgroup TIM_LL_EC_ONEPULSEMODE One Pulse Mode
  * @{
  */
#define LL_TIM_ONEPULSEMODE_SINGLE             TIM_CR1_OPM           /*!< Counter stops counting at the next update event */
#define LL_TIM_ONEPULSEMODE_REPETITIVE         0x00000000U          /*!< Counter is not stopped at update event */
/**
  * @}
  */

Hope this helps.

Best Regards,
Bas

ASELSTM, 2021-03-29:
Hi @basvloon, 

Thank you for this report. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release.

We cannot share a date for the moment. So stay tuned and thank you once more for your contribution and for your patience.

With regards

ASELSTM, 2022-03-11:
Hi @basvloon,

This issue has been fixed in the frame of version v1.10.0 of the STM32CubeH7 published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM