STMicroelectronics/STM32CubeH7 #129: Question: USART_IRQ.TXFT

rxa1031, 2021-03-27:
Hello,

I am looking at function `HAL_UART_Transmit_IT()` implementation. I observe below call:

https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L1332

However I do not observe a code piece associated with USART_ISR.TXFT (i.e. `USART_ISR_TXFT`) being available with the `HAL_UART_IRQHandler()` implementation.

I only observe the `USART_CR3_TXFTIE` being Cleared within function UART_TxISR_8BIT_FIFOEN()
https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L4080

I am unaware why the latest code implementation SETs  USART_CR3.TXFTIE bit, but never uses its corresponding Interrupt TXFT present with USART_ISR register.

Please help me understand the reason for this.

Thanks,
Rajeev

rxa1031, 2021-03-29:
Some other observations:

The STM32CubeH7 code calls `HAL_UARTEx_EnableFifoMode()`. The code with the function SETs the `USART1_CR1_FIFOEN bit`.

With help of _ARM/KEIL ULINK Pro debugger_ I am observing that `USART_ISR_TXFT` bit is set as a result of this.
[image]

I observe that on calling `HAL_UART_Transmit_IT()` `bit USART_CR3_TXFTIE` is SET. This triggers the `USART_ISR_TXFT` and other interrupt.

I am **concerned** why the `UART_TxISR_8BIT_FIFOEN` is executed with help of `USART_ISR_TXFNFE interrupt` and not with help of  `USART_ISR_TXFT interrupt` when FIFO use is enabled?
https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L2497-L2503

Please help me understand above.

Thanks,
Rajeev

rxa1031, 2021-03-29:
As per my understanding, there are a maximum of 16 TX FIFO buffers.
If TX FIFO is empty `USART_ISR_TXFT interrupt` may have been used to Fill all 16 TXFIFO buffers (if Number of TX bytes were >= 16)
The coder may have set the `USART_CR3_TXFCTCFG = 0b100` (i.e. TXFIFO reaches 7/8 of the depth, i.e.  16 - 16 * 7 / 8 = 2 bytes (or less) are left with the TX FIFO). In this case 14 bytes (if available for Transmission) can be added to the TX FIFO using the `USART_ISR_TXFT interrupt`.

ASELSTM, 2021-04-06:
Hi @rxa1031,

It is true that UART IRQ handler processing is not using TXFT flag, even if UART transmission is done with FIFO mode enabled. Here is how it is intended to work :

TXFTIE is enabled for starting a transmit operation. Since no data is present in Tx Fifo, an interruption is raised with the flag TXFNF that is expected to be 1. This will then trigger the execution of  `UART_TxISR_8BIT_FIFOEN` which will lead to filling the TX Fifo until all data are copied in TX Fifo, or until the TX Fifo is full.

Then, when bytes will be sent, the TX FIFO will be emptied until the TXFT flag is set and a new interrupt is raised.  In this case, both TXFT and TXFNF flags are set (TXFNF could not be 0 if TXFT is 1). This will trigger the execution of `UART_TxISR_8BIT_FIFOEN`.  And so till all data bytes are transmitted. 

In other terms, is TXFT flag is set to 1 then so is the TXFNF flag.  So that interruption is raised and `UART_TxISR_8BIT_FIFOEN` is executed then.

Testing on the TXFNF, allows to call continuously the TxISR without worrying about the source of the interruption. It allow then to reload the FIFO as soon as possible.

Hope this would help you.

With regards,

rxa1031, 2021-04-07:
Hello @ASELSTM ,

Please let me know whether my below understandings are correct.

With TXFT interrupt not being tested in USART IRQ Handler, one might enable TXFNFIE (without enabling TXFTIE) and still have the same result.

It appears that only use of enabling FIFO use is to allow at least the TX to write multiple bytes to Tx FIFO without having to wait for the transmission of every byte (as in the case when Tx FIFO is disabled).

Also from response RXFIFO related with point 3 of #130 , I understand that even if FIFO is enabled, it will only help queue the UART Receiver received bytes, and would trigger RXFNFE interrupt for every received byte.

Thanks,
@rxa1031

ASELSTM, 2021-04-08:
Hi @rxa1031,

- The final behavior of your application will actually be the same. However, activating TXFTIE will reduce the number of interruptions. Indeed, once activated interruption will occur once the threshold reached otherwise interruptions occur each time a byte is transferred. The activation, or not, of interruption related to the Threshold will mainly depend on your application and on the use case.

- Indeed, it is one of the uses of the FIFO feature.  The FIFO helps also, when RXFTIE and TXFTIE are set,  to release the CPU by reducing the number of interruptions generated. The FIFO is also useful for high speed transfer.

- The Tx FIFO and Rx FIFO behavior are obviously the same. As you have said, FIFO will queue the UART received data and RXFNFE interrupt will be trigged at each received byte unless you activate the threshold interruption (RXFTIE) so that RXFT flag is set once threshold is reached and an interruption will be trigged.

I hope you find this answer helpful.

With regards,

closed 2021-04-08 by rxa1031