STMicroelectronics/STM32CubeH7 #131: FatFs Application for STM32H745I-DISCO not working properly

simosilva, 2021-03-30:
**SETUP**:
  The board : STM32H745I-DISCO
  IDE: Keil uVision 5.28a

**Describe the bug**:
The Application, once compiled both for CM4 and CM7 and flashed on the discovery, does not work properly.

The terminal prints the following lines:
```
[CM7]:  Creating FileSystem...
[CM7]:    FileSystem Created successfully!
[CM4]:       starting FatFs operation....
[CM7]:      writing on 'cm7.log'
[CM4]:      opening 'cm4.log' for writing...
[CM7]:  'cm7.log' '34' bytes written...
[CM4]:        Error!!
[CM7]:  Error!!
```

Checked the two codes and seem identical, the error came when CM4 try open the file with the f_open (line 147 of the main.c).
Please resolve this bug thanks.

**How To Reproduce**
simply download, compile and flash the firmware from https://github.com/STMicroelectronics/STM32CubeH7/tree/v1.9.0/Projects/STM32H745I-DISCO/Applications/FatFs/FatFs_Shared_Device ( v1.9.0 tagged repo)

Thank you!

RKOUSTM, 2021-04-01:
Hi @simosilva,
 
Thank you for your contribution. The issue you pointed out has been confirmed.  An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-04-02:
ST Internal Reference: 104230

ALABSTM, 2023-04-04:
Hi @simosilva,

Thank you again for having reported this point. It has been fixed in the frame of version **1.10.0** of this firmware.

With regards,

closed 2023-04-04 by ALABSTM