STMicroelectronics/STM32CubeH7 #132: Wrong HSE_VALUE in NUCLEO-H743ZI LwIP Application

kkrambo, 2021-04-08:
The HSE_VALUE should be 0x08000000 instead of 0x25000000.  (By default, the HSE is sourced from the fixed 8 MHz MCO from the ST-LINK microcontroller on the NUCLEO boards.)  This bug might be benign for the lwIP application as it is.  But it caused me grief when I updated the project to add a USART feature because it caused the baud rate to be incorrect.

https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Inc/stm32h7xx_hal_conf.h#L97

ASELSTM, 2021-04-12:
Hi @kkrambo,

First we would like to thank you for your contribution. The issue you pointed out is confirmed. In fact, the HSE_VALUE value should be 8000000 instead of 25000000. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

ASELSTM, 2021-04-12:
ST Internal Reference: 104729

ASELSTM, 2022-03-11:
Hi @kkrambo,

This issue has been fixed in the frame of version v1.10.0 of the STM32CubeH7 published recently. Please allow me then to close this thread.

With regards,