STMicroelectronics/STM32CubeH7 #133: Query: Declaring a Global variable in External MRAM / SRAM and simultaneously initializing it within one C language statement.

rxa1031, 2021-04-14:
Hello,

Used tools and embedded software packs:
STM32CubeMX version 6.2.1
STM32CubeH7 version 1.9.0
Keil compiler version V6.16 (Available with MDK534.EXE)

Requirements:
1. Declare a Global variable with a known value in External SRAM:
- NOINIT section
- INIT Section

My understand is that the only difference between a NOINIT and an INIT section is that the memory region marked with INIT is to be updated by function `SystemInit` such that all the bytes in that region are initialized with a `value of 0`. Where as the memory region declared as NOINIT is not cleared.

Peculiar issue:
1. Even though the STM32CubeMX tool is used to inform the size / bytes and other details of the External SRAM, the tool does not add any section / details to the SCT file.
2. Where to add changes in system_stm32h753xx.s or system_stm32h7xx.c so as to execute the System Clock and other clocks Init, GPIO Init, and FMC Init before executing `SystemInit` function.
3. Should the External SRAM details be also added to `Options for Target'Project' -> Target -> Read/Write Memory Areas -> RAM1, RAM2 and RAM3`?
4. Unable to declare global variables like below which reside in External NOINIT and INIT areas, without causing a code crash
`char ch = 'x' __attribute__( ( section( "EXT_INIT_RW_RAM_SECTION" ) ) );`
5. I am not sure what should be the content of the SCT file for the EXTERNAL SRAM (both NOINIT and INIT areas)

Please help.

Thanks,
@rxa1031

ALABSTM, 2021-04-21:
Hello @rxa1031,

As said in #134, you can take the [ExtMem_Boot](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H750B-DK/Templates/ExtMem_Boot) template of the **STM32H750B-DK** _Value Line_ board as a basis to adapt the `SystemInit()` function of your application. Further details are given by application note [AN5188](https://www.st.com/resource/en/application_note/dm00514974-external-memory-code-execution-on-stm32f7x0-value-line-stm32h750-value-line-stm32h7b0-value-line-and-stm32h730-value-line-mcus-stmicroelectronics.pdf).

Regarding aspects, related to **Cube MX**, please submit your questions and requests to the **ST Community**, in the dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx).

With regards,

ALABSTM, 2021-05-24:
Hi @rxa1031,

Thank you again for this report. Please allow me to close this point and continue the discussion in #145.

With regards,

closed 2021-05-24 by ALABSTM