STMicroelectronics/STM32CubeH7 #134: Query: Tweaking __main to call System Clock Configure, Peripheral Configure and FMC Initialization before calling Zero Initialization of memory

rxa1031, 2021-04-16:
Hello,

I am currently unable to declare initialized Global Variables in External SRAM because of the limitations with __main function (available with Keil compiler version V6.16 ) code availability and use of STM32CubeMX (version 6.2.1). The generated code is using the HAL drivers for System Clock, Peripherals and FMC.

I had earlier informed in one of my queries that when FMC is configured with STM32CubeMX, the tool does not embed details of the external SRAM to the SCT file that is used by Keil Compiler.

I read the document [Application Startup](https://www.keil.com/support/man/docs/armclang_intro/armclang_intro_asa1505906246660.htm) and understand that if I want to declare an Initialized Global Variable in external FMC SRAM, then this can only be done by tweaking the __main or preceding __main execution with a function call / execution that:
1. Initializes Clocks
2. Initializes at least the the FMC Pins
3. Initializes the FMC communication (signal settings).

On reviewing the function `SystemInit`, which is called before __main, I observe that the function only uses Microcontroller registers, and assigns constant values to the used registers, and I understand the reason for why variables are not used.

Current understanding:
1. I understand that code of __main is available in some Keil provided library and hence cannot be tweaked. So the available option would be to call / execute a user function (which does some user required actions) before executing __main .
2. The STM32CubeMX provides function like SystemClock_Config(), PeriphCommonClock_Config() and MX_FMC_Init() which use at least some local variables / structures, and hence these functions cannot be called before calling __main.

Considerations:
1. Not everyone who uses Keil and uses the SBM32 microcontrollers will be able to correctly tweak the  the System Clock,  Peripheral Configuration and FMC Initialization functions.
2. The STM32CubeMX tool will overwrite any tweaks that are made to above functions or any part of the code which are not present in the user section.

Requests:
1. I would request you to  please provide a custom (weak) function (called from within __main) that may be modified by user.  This function call should exist:
A. before the call that initializes global variables and zeroes the ZI section
B. Exist after the stack is created.
2. If above is not possible, then help me with a tool that helps remove local variables and embed fix values in code, so that a user can create a Custom function from a stable function (generated by STM32CubeMX or provided by Keil) and executes the custom function before __main?
3. If even the above is not possible then please help me with a LL Driver for the three functions. the LL Driver must not use any variables and should be available / generated through the STM32CubeMX, when a future version of STM32CubeH7 is released.

Please help me with your inputs.

Thanks,
@rxa1031

rxa1031, 2021-04-16:
Hello,

Please also help provide a user code section just before `LDR     R0, =__main` in file `startup_stm32h753xx.s.`

Thanks,
@rxa1031

rxa1031, 2021-04-16:
Hello,

When comparing the STM32CubeH7 / STM32CubeMX implementation with the Keil Provided CMSIS implementation, I found some inputs which I want to share with you.

As per ARM / Keil, it is expected, all the memory used by the application is enabled before calling __main. In the CMSIS standard, the `SystemInit()` function is the place for this.
I obviously `macro DATA_IN_ExtSRAM` and code for SRAM in `ystem_st32f4xx.c` file, which accompanies the STM32F4xx_DFP.
```void SystemInit(void)
{
  /* FPU settings ------------------------------------------------------------*/
  #if (__FPU_PRESENT == 1) && (__FPU_USED == 1)
    SCB->CPACR |= ((3UL << 10*2)|(3UL << 11*2));  /* set CP10 and CP11 Full Access */
  #endif

#if defined (DATA_IN_ExtSRAM) || defined (DATA_IN_ExtSDRAM)
  SystemInit_ExtMemCtl(); 
#endif /* DATA_IN_ExtSRAM || DATA_IN_ExtSDRAM */

  /* Configure the Vector Table location add offset address ------------------*/
#ifdef VECT_TAB_SRAM
  SCB->VTOR = SRAM_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
#else
  SCB->VTOR = FLASH_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
#endif
}```

I also came across below comments under `release version 1.1.0` present with document [Release Notes for STM32F7xx CMSIS 1.2.1](http://mazsola.iit.uni-miskolc.hu/~drdani/docs_arm/st/sw/Libraries/CMSIS_Device_STM32F7xx_1.2.1/Release_Notes.html)
  
**system_stm32f7xx.c/.h files** 
- 	Remove external memories configuration from the system_stm32f7xx.c common file (moved to Template Projects)

May ST please help me understand the reason for above, and how do I implement my requirement without having any adverse impact from regeneration of Code (using STM32CubeMX) and ensuring that my team does not have to bother about adding any tweaks again to the project.

Please help.

Thanks,
Rajeev

ALABSTM, 2021-04-21:
Hello @rxa1031,

The use-case you are describing is typical of the **STM32H750B-DK** _Value Line_ board. In order to enable the external memory access during start-up, you have to adapt the `SystemInit()` function as you suggested. Please have a look to the [ExtMem_Boot](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H750B-DK/Templates/ExtMem_Boot) template.

As you can notice in the `system_stm32h7xx.c` file, the `SystemInit()` contains the call below.
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Projects/STM32H750B-DK/Templates/ExtMem_Boot/Src/system_stm32h7xx.c#L227-L229

The `SystemInit_ExtMemCtl()` function is defined within the same file.
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Projects/STM32H750B-DK/Templates/ExtMem_Boot/Src/system_stm32h7xx.c#L365

You simply have to port these into the `system_stm32h7xx.c` of your application and adapt the `SystemInit_ExtMemCtl()` according to the GPIOs used to communicate with your external memory. I hope this helps.

Regarding aspects, related to **Cube MX**, please submit your questions and requests to the **ST Community**, in the dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx).

With regards,

rxa1031, 2021-05-07:
Hi @ALABSTM ,

I believe the STM32CubeH7 project can provide mechanism using conditional compile code to help implement what I wanted (i.e. initialize Clocks, configure FMC related GPIOs, and configure FMC SRAM) before a call to __main() is made.

I coded RCC / Clock related functions, FMC GPIO configuration and FMC configuration functions_ which only use the Microcontroller registers and constant values.

I also coded my own `Reset_Handler()` function.

I am now able to declare global variables in external SRAM / MRAM which can be initialized while declaring.

I am now able to declare both UNINIT and Initialize with Zero (or) known value in External SRAM

Attached code for reference:

[FMC-SRAM_C_Code_as_TXT_File.TXT](https://github.com/STMicroelectronics/STM32CubeH7/files/6441150/FMC-SRAM_C_Code_as_TXT_File.TXT)

[SCT_FILE_EXTENSION_CHANGED_TO_TXT.TXT](https://github.com/STMicroelectronics/STM32CubeH7/files/6441161/SCT_FILE_EXTENSION_CHANGED_TO_TXT.TXT)

Sample Global Variables declaration for testing:
```
#define dExternamUninitializedMRAM_Segment      __attribute__( ( section( ".bss.EXT_NO_INIT_RW_SRAM_SECTION" ) ) )

#define dExternamZeroInitializedMRAM_Segment      __attribute__( ( section( ".EXT_ZERO_INIT_RW_SRAM_SECTION" ) ) )

uint8_t au8ExternalMRAM_ZeroInitializedReadAndWriteBuffer[dExternalMRAM_TestBufferSize] dExternamZeroInitializedMRAM_Segment = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10 };

uint8_t au8ExternalMRAM_UninitializedReadAndWriteBuffer[dExternalMRAM_TestBufferSize] dExternamUninitializedMRAM_Segment;
```

Regards,
Rajeev

ALABSTM, 2021-05-24:
Hi @rxa1031,

Thank you again for this report. Please allow me to close this point and continue the discussion in #145.

With regards,

closed 2021-05-24 by ALABSTM

ALABSTM, 2021-06-17:
Duplicate of #145