STMicroelectronics/STM32CubeH7 #138: MX25LM51245G timing is not respected

GiulioDallaVecchia, 2021-04-21:
**Describe the set-up**
 * STM32H735G-DK
 * STM32CubeIDE v1.6.0

**Describe the bug**
Erasing memory timeout is not respected. 

**How To Reproduce**
1. Create a new project 
2. Import stm32h735g_discovery_ospi module
3. Import mx25lm51245g.c as driver
4. Try to call BSP_OSPI_NOR_Erase_Chip() and check that an error is returned

**Additional context**
You should to modify MX25LM51245G_AutoPollingMemReady function passing a timeout value depending of the operation that the user want to handle.

RKOUSTM, 2021-04-29:
Hi @GiulioDallaVecchia,

First we would like to thank you for your contribution.  In order to allow a better analysis of your problem, would you please share your project for further analysis.

Thank you once more for your contribution.

With regards,

RKOUSTM, 2021-05-12:
Hi @GiulioDallaVecchia,

I hope you are fine. Any new feedback regarding this issue?

With regards,

GiulioDallaVecchia, 2021-05-13:
Hi @RKOUSTM,
I apologize, let me few days in order to prepare a project to show you the problem.

With regards,

GiulioDallaVecchia, 2021-05-17:
Hi @RKOUSTM,

RKOUSTM, 2021-05-25:
Hi @GiulioDallaVecchia, 

Thank you for your contribution. Your report will be forwarded to our development team. We will get back to you as soon as we have more details.

Thank you once more for this contribution.

With regards,

RKOUSTM, 2021-06-18:
Hi @GiulioDallaVecchia,

Thank you for you are contribution.  We tried to reproduce the issue you described, but we couldn't notice the problem. 

To address you need, we have proposed a modification in the octospi.c file of our BSP example running on STM32H735G-DK. This update is proposed to call  the MX25LM51245G_AutoPollingMemReady() function in order to verify if the memory is ready or not. 

Please find attached a modified version for octospi.c file that you want the integer in the BSP example.
[Test_OSPI_NOR_Erase.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6677084/Test_OSPI_NOR_Erase.zip)

I hope that the modification we have provided you with has been helpful. Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2021-06-18 by RKOUSTM