STMicroelectronics/STM32CubeH7 #139: Licensing of Examples

rmspacefish, 2021-04-27:
Hello there,

Thanks for providing the application examples for lwIP. They were very helpful for me as I am trying to port them to the RTEMS OS. However, there is still a small issue which makes it difficult to release the port as part of RTEMS which generally requires an OpenSource license like a BSD license.

The example applications appear to be licensed differently than the HAL library ( for example: https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS) . It appears to be licensed under the ST Ultimate Liberty license SLA0044 instead of the BSD 3-Clause. Point 5 of SLA0044 is especially problematic. My port is based on your example applications for lwIP, but was heavily modified to be usable for the RTEMS OS. Would it be possible to license these files under BSD 3-Clause, similarly to the HAL library? Thanks a lot in advance.

Kind Regards
Robin

ALABSTM, 2021-05-24:
Hi @rmspacefish,

Please forgive my delayed reply. I forwarded your request to both our development and quality teams as soon as you posted it. The point is still being discussed. I will try to push to have an answer as soon as possible. I will keep you informed.

Thank you for your patience and comprehension and thank you again for having tackled the subject.

With regards,

ALABSTM, 2021-06-09:
Hi @rmspacefish,

I hope you are fine. Your request is still under discussion.

May I ask whether the ported code is intended to run on STMicroelectronics hardware exclusively once integrated into the RTEMS OS project? May I also ask for more precisions about the "vendor device restrictions" you mentioned?

Thank you,

rmspacefish, 2021-06-09:
Hi,

Thanks for the reply.

The  ported code with the licensing issues glues the lwIP code to the STM32 and the used PHY so this part will be run on STMicorelectronics hardware exclusively. I asked in the RTEMS mailing list about more details concerning the vendor device restrictions and might be able to supply more information soon.

Kind Regards
Robin

gedare, 2021-06-09:
Vendor device restrictions means a clause related to running the code only on STMicroelectronics hardware. Such a clause can be acceptable, but a true permissive OSI-approved license (e.g., 2- or 3-BSD) would be preferred for simpler compliance reasons.

rmspacefish, 2021-07-15:
Hello,

Has there been any progress on the issue?

Kind Regards
Robin

ALABSTM, 2021-07-15:
Hello @rmspacefish,

Discussions are still ongoing. I will keep you informed. Thank you for your comprehension.

Kind regards,

ALABSTM, 2021-08-02:
Hello @robamu and @gedare,

After long discussions and due to a number of considerations, we decided to keep these two files under `SLA0044` license as the rest of our demos and applications. We really wished to provide you with a positive answer, but could unfortunately not. Best of luck for your project and thank you for your comprehension.

With regards,

robamu, 2021-08-03:
That's unfortunate. Thanks for your efforts nonetheless.

ALABSTM, 2021-08-05:
Hi @robamu,

You are welcome and thank you for your comprehension again. Please allow me to close this issue now.

With regards,

closed 2021-08-05 by ALABSTM

joelsherrill, 2021-08-05:
Is STM adamant about paragraph 5 of SLA0044? Paragraph 4 requires use on only STM devices which is acceptable. The intent of paragraph 5 is unclear. 

It could be read that software under this license cannot be distributed with software using a viral free software license or that it cannot be distributed with software under any open source license.  I honestly cannot tell what STM wants. 

We want people building products using STM devices to be able to use this code with RTEMS but paragraph 5 is written in a way that it is impossible to know what the intended action by someone would be to be compliant to that license. All we want to do is include the driver with a port of lwip that runs on RTEMS without going against STM's desired use cases.

@gedare and I are core developers from RTEMS with decades of open source and free software experience between us. We would be happy to discuss this privately if needed to help clarify the issues.