STMicroelectronics/STM32CubeH7 #141: OpenAMP_RTOS_PingPong does not work with new MX Version

JSCLadon, 2021-05-05:
**Setup**
 * Board: STM32H747i-DISCO
 * IDE: STM32CubeIDE Version: 1.6.0.202101291314
 * STM32CubeMX: Version: 6.2.1-RC2

**Description of Bug**
The example is fairly outdated: works with CMSIS V1 and does not support CubeMX.
By porting it to the new hierarchical project structure in CubeIDE in combination with CubeMX and the [Release Notes](https://www.st.com/resource/en/release_note/dm00107607-stm32cubemx-release-621-stmicroelectronics.pdf), the example does a clean compile, but there is no communication between the two cores.

**How To Reproduce**

1. download and unpack the [project](https://ladon-my.sharepoint.com/:u:/g/personal/johannes_schnitzler_ladon-energy_com/EUmhwvs8n49ElH_UpGwxe8wB9Esgnjqz8hu71v7wnAS_4Q?e=1O8TOi)
2. Compile & debug
3. Core M4 waits in an endless loop to get access to the shared memory

Thanks a lot for taking care of the problem. 
Examples are generally very helpfull to start with a new project. Therefore it is worth the effort to keep the examples updated.
I would love to contribute, but I don't manage to get it to work properly.

Johannes

RKOUSTM, 2021-05-12:
Hi @JSCLadon,

First, allow me to thank you for your contribution. There are no plans to add .ioc files for STM32H7 products. Unfortunately, we don't treat aspect related to CubeMx tool in our GitHub repositories. Please allow me then to redirect you to the ST Community. You can report this issue in the CubeMX related [section](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros). There you shall find people who will address your request and whom you will hopefully find a solution with.

Thank you for your comprehension.

Thank you again for your report. We are looking forward to reading from you again.

With regards,

JSCLadon, 2021-05-12:
Hi @***@***.***>

Thank you very much for your reply. I appreciate, that you took care of my request.
Although I must say, that it is not very helpful.
I have already addressed the issue in several other forums:
https://community.st.com/s/question/0D53W00000kvuIXSAY/communication-between-cores-with-freertos-on-stm32h747idiscovery-how-can-i-adjust-the-example-freertosampdualrtos-which-is-designed-for-the-stm32h745dicovery-board
https://community.st.com/s/question/0D53W00000hNgU7SAK/openamp-freertos-pingpong-demo-isnt-working-on-the-nucleoh745ziq-with-the-m4-getting-hung-waiting-for-the-resource-table-to-be-configured-by-the-m7
and even
https://community.st.com/s/feed/0D53W00000m2AACSA2

But with very little response.
We have now developed it ourself, but since I paid for it, I can not put it into the community for free, but try to sell it to the other developers, who are interested in getting the examples in MX compatible format. There are plenty of requests on this to be found in the different forums.

I am a big fan of the new STM32 Cube platform. I strongly believe, that this is the right way to go to adapt new MCU into R&D departments. The examples you provide could really be helpful, because they are covering a lot of the important new development trends.

But I do not understand, why you do not support your own tools better by making all the examples available in MX compatible format in the first place!

Please take this as my personal opinion to untap new room for improvement. English is a foreign language to me. If I did not pick the right words and my email sound rude or unpolite, I apologize for that.

Mit freundlichen Grüßen / Kind regards

Johannes Schnitzler
Geschäftsführer

Ladon Energy GmbH
Tiergartenstrasse 164A
30559 Hannover
+49 511 533 666 1
+49 172 8 142 143
***@***.***

coledd, 2021-05-12:
Johannes, I could not agree more. I've been waiting for ST to connect the
examples to CubeMX for years. It is really odd to me that ST does not eat their
own dog food :)

-Chris

On Wed, 2021-05-12 at 02:23 -0700, Johannes wrote:
> But I do not understand, why you do not support your own tools better by
> making all the examples available in MX compatible format in the first place!