STMicroelectronics/STM32CubeH7 #142: Wrong pin config in stm32h747i_eval_nor.c

trinhnhatnam, 2021-05-05:
This is in the current code:

`  /* GPIOC configuration */
  gpio_init_structure.Alternate = GPIO_AF9_FMC;
  gpio_init_structure.Pin   = GPIO_PIN_6 | GPIO_PIN_7;
  HAL_GPIO_Init(GPIOC, &gpio_init_structure);

  /* GPIOD configuration */
  gpio_init_structure.Alternate = GPIO_AF12_FMC;  //add GPIO_PIN_7
  gpio_init_structure.Pin   = GPIO_PIN_0 | GPIO_PIN_1 | GPIO_PIN_4 | GPIO_PIN_5|GPIO_PIN_7|\
                              GPIO_PIN_8 | GPIO_PIN_9 | GPIO_PIN_10 | GPIO_PIN_11 |\
                              GPIO_PIN_12 | GPIO_PIN_13 | GPIO_PIN_14 | GPIO_PIN_15;
  HAL_GPIO_Init(GPIOD, &gpio_init_structure);`

What's wrong is PC6 and PC7 are not used for FMC, so they need to be removed, and clock for GPIOC doesn't need to be turned on. Then GPIO_PIN_6 needes to be added to GPIOD init. So replace the above code with this:

`  /* GPIOD configuration */
  gpio_init_structure.Alternate = GPIO_AF12_FMC;
  gpio_init_structure.Pin   = GPIO_PIN_0 | GPIO_PIN_1 | GPIO_PIN_4 | GPIO_PIN_5| GPIO_PIN_6| GPIO_PIN_7|\
                              GPIO_PIN_8 | GPIO_PIN_9 | GPIO_PIN_10 | GPIO_PIN_11 |\
                              GPIO_PIN_12 | GPIO_PIN_13 | GPIO_PIN_14 | GPIO_PIN_15;
  HAL_GPIO_Init(GPIOD, &gpio_init_structure);`

ASELSTM, 2021-05-10:
Hi @trinhnhatnam,

Referring to the [datasheet](https://www.st.com/resource/en/datasheet/stm32h747xi.pdf) , the PC6 and PC7 pins are actually used for FMC as shown below : 

[image]

In fact, PC6 is linked to the `FMC_NWAIT` which could also be mapped on PD6. Similarly for PC7, which is linked to `FMC_NE1` and could also be mapped on PD7. Therefore, the use of PC6 and PC7 is not erroneous.

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2021-05-10 by ASELSTM

ASELSTM, 2021-05-11:
ST Internal Reference: 106824