STMicroelectronics/STM32CubeH7 #145: Request: Please modify following functions SystemClock_Config(), PeriphCommonClock_Config(), and MX_FMC_Init() along with its called functions so as not to use variables

rxa1031, 2021-05-17:
Hello,

I had share with you my concerns #133 and #134

I had shared the code changes that I made to STM32CubeH7 version 1.9.0 (under #134 )

If developers (who code the application / firmware that execute on microcontroller) depend solely on STM32CubeH7 provided drivers, then they will never be able to fix the issues that I had talked about. Also not everyone is able to tweak the drivers, it takes good understanding of the Datasheet, User Manal, Protocol etc.

Hence, ST Microelectronics as the chip manufacturer and as the (sample) driver provider could help above people by tweaking functions like SystemClock_Config(), PeriphCommonClock_Config(), and MX_FMC_Init() along with its called functions so that they do not use variables. thus If the coder only modifies the Reset_Handler() function body, the rest changes will have already been taken care, and the developer / coder does not have to know how to resolve the issue (as it will never appear).

A part of help for this can also come from the STM32CubeMX developers.

Please share your views.

Thanks,
Rajeev

ALABSTM, 2021-05-24:
Hi @rxa1031,

Thank you very much for your analysis and your proposals. Your request has been forwarded to our development teams. We will get back to you as soon as we receive their feedback.

With regards,

ALABSTM, 2021-06-17:
Hi @rxa1031,

Thank you again for your analysis and your proposals. According to our development teams, the use of external memories remains **project-specific**. Even the GPIO pins interfacing with an external memory are project-specific.

This is why, as you noticed about the STM32Cube**F7** firmware, the external memories configuration code has been moved from the `system_stm32f7xx.c` common file to the different project templates (with board-specific adaptations). The same update may be expected for the STM32Cube**F4** firmware.

As for developing an application based on an external memory, with the several examples provided for several boards, particularly the [STM32756G_EVAL](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32756G_EVAL/Examples/FMC) board, users have enough material to start from.

Thank you again for your proposals. Please allow me to close this issue now.

With regards,

closed 2021-06-17 by ALABSTM