STMicroelectronics/STM32CubeH7 #146: Example: MDMA_RepeatBlock_Rotation - Readme says 90° rotation but not done in source code

Counterfeiter, 2021-05-21:
**Describe the set-up**
 * I have done just a source review

**Additional context**
I am looking for a way to transpose a matrix really fast. It is the same operation as rotate a image 90°. 
I would like to use the MDMA for this. I tought I found the solution with your example, but looking deeper in the example description the 90° rotation with the MDMA isn't done.

The question/issue is, is it possible with the MDMA and it's not done in your example (source code issue) or is it not possible with the MDMA (readme issue)?

Thx for clarifying this (and fixing the example).

Best regards

Sebastian

RKOUSTM, 2021-05-21:
Hi @Counterfeiter,

Thank you for your contribution. The issue you pointed out has been confirmed. An internal tracker has been logged and an update in the `readme.txt` will be made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-05-21:
ST Internal Reference: 107449

RKOUSTM, 2021-12-14:
Hi @Counterfeiter,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.1 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,