STMicroelectronics/STM32CubeH7 #147: STMCUBEMX STM32H7A3 USB init bug

xRadx, 2021-05-26:
**Set-up**
 * The board: nucleo H7A3
 * IDE: STM32CubeIDE/MX

**the bug**
Code generator does not generate line HAL_PWREx_EnableUSBVoltageDetector() which causes USB device is not being detected/USBH_UserProcess is not being executed. 
For nucleo H743 the line is generated and USB works.

RKOUSTM, 2021-05-27:
Hi @xRadx,

Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us the .ioc file that you are used to reproduce this issue. Thank you again for your contribution.

With regards,

RKOUSTM, 2021-06-17:
Hi @xRadx, 

I hope you are doing well. Any new comments to move this discussion forward?

with regards,

xRadx, 2021-06-18:
[test1.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6675221/test1.zip)
I am sorry for late response.
file attached

RKOUSTM, 2021-06-22:
Hi @xRadx,

Thank you for your contribution.  After investigation, the point you reported is related to CubeMX tool  and not directly related to the firmware published in this repository. 

Please find attached an .ioc file for STM32H7A3 Nucleo board using STM32CubeMX v6.2.1. In fact, an update on the pinout and system clock configuration is proposed to solve your reported problem.  Now, the `HAL_PWREx_EnableUSBVoltageDetector()` is generated inside the `HAL_HCD_MspInit()` function in the `usbh_conf.c` file. 

[Test1.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6692449/Test1.zip)

I hope the .ioc file I provided you with was helpful. In general terms, we don't treat aspects related to **CubeMX** tool at GitHub level, they are rather treated at  [ST Community](https://community.st.com/s/topiccatalog) level. As this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (**CubeMX** tool), please allow me to close it now. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2021-06-22 by RKOUSTM