STMicroelectronics/STM32CubeH7 #148: H745ZI-Q SPI using DMA is not working using Stm32CubeMx

ahmedwahdan, 2021-05-26:
**Describe the set-up**
 * The board : H745ZI-Q
 * IDE: StmCubeIde

**Describe the bug**
SPI using DMA is not working using Stm32CubeMx, call back is not triggered, no errors.

**How To Reproduce**
Using SPI with DMA generated code using StmCubeMx

RKOUSTM, 2021-05-27:
Hi @ahmedwahdan,
 
Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us the .ioc file that you are used to reproduce this issue. Thank you again for your contribution.

With regards,

RKOUSTM, 2021-06-17:
Hi @ahmedwahdan,

I hope you are doing well. Any new comments to move this discussion forward?

with regards,

RKOUSTM, 2021-06-22:
Hi @ahmedwahdan,

Please allow me to close this issue. You can reopen it if you have anything else relevant to add regarding this topic.

With regards,

closed 2021-06-22 by RKOUSTM