STMicroelectronics/STM32CubeH7 #152: Not able to flash any code on NUCLEO-H745ZI-Q using STM32CubeIDE

sourabhmisal, 2021-06-02:
I am trying to flash a basic LED program on NUCLEO-H745ZI-Q, recently purchased, for the past 2 days. 
Console output shows that download is successful, but the LD1 pin (PB0) is not blinking.
I am using STM32CubeIDE version 1.5.1 on Ubuntu 20.04 LTS
I have generated the code with default peripheral settings and just added
```
HAL_GPIO_WritePin(GPIOB, GPIO_PIN_0, GPIO_PIN_RESET);
HAL_Delay(1000);
HAL_GPIO_WritePin(GPIOB, GPIO_PIN_0, GPIO_PIN_SET);
HAL_Delay(1000);
```
in the while loop, but unsuccessful in output.. could anyone guide me out?
Console output when trying to run the basic flash program:

```STMicroelectronics ST-LINK GDB server. Version 5.7.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled
        InitWhile                  : Enabled

Info : default port : 7184
Info : Remote address: 127.0.0.1
Info : stlinkserver already running, exit
Waiting for debugger connection...
Debugger connected
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.6.0-RC3                  
      -------------------------------------------------------------------

ST-Link Server is running on port : 7184
Info : default port : 7184
Info : Remote address: 127.0.0.1
Info : stlinkserver already running, exit
Info : default port : 7184
Info : Remote address: 127.0.0.1
Info : stlinkserver already running, exit
ST-LINK SN  : 003E00333137511039383538
ST-LINK FW  : V3J7M2
Board       : NUCLEO-H745ZI-Q
Voltage     : 3.27V
SWD freq    : 24000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x450
Revision ID : Rev V
Device name : STM32H7xx
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7/M4

Memory Programming ...
Opening and parsing file: ST-LINK_GDB_server_Dzv7j5.srec
  File          : ST-LINK_GDB_server_Dzv7j5.srec
  Size          : 21756 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sector 0
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:03.100

Verifying ...

Download verified successfully 

Info : default port : 7184
Info : Remote address: 127.0.0.1
Info : stlinkserver already running, exit
Debugger connection lost.
Shutting down...
```

RKOUSTM, 2021-06-04:
Hi @sourabhmisal,

Thank you for your contribution.  In order to allow a better analysis of this problem, could you please confirm that when you use one of our applications, the issue is systematically reproducible on your side?

Thank you again for your contribution.

Best regards,

sourabhmisal, 2021-06-11:
I will try out one of the examples mentioned, but i found out a way to use M7 core alone by commenting out this piece of code generated by STM32CubeIDE

main.c file

Boot_Mode_Sequence_1
```
//  timeout = 0xFFFF;
//  while((__HAL_RCC_GET_FLAG(RCC_FLAG_D2CKRDY) != RESET) && (timeout-- > 0));
//  if ( timeout < 0 )
//  {
//  Error_Handler();
//  }
```

when i was trying to debug, the code gets stuck in error handler. I think it waits for M4 core to run and waits for Flag to set. Since i wanted to use M7 core for now, by commenting boot mode sequence, i was able to get things done on M7 core. I will explore the M4 and dual core functionality soon enough.

closed 2021-06-11 by sourabhmisal