STMicroelectronics/STM32CubeH7 #157: Enhancement: Provide a function that is capable to storing multiple CAN messages to CAN TX FIFOs before updating TXBAR

rxa1031, 2021-06-09:
Hello,

I am trying to find a function that helps first find all available TX FIFOs and later fills them with (Classic) CAN Data and Extended IDs before updating a value to _hfdcan->Instance->TXBAR_. This code will be helpful to me in the long run.

There are only two functions in file _stm32h7xx_hal_fdcan.c_ that modify value of _hfdcan->Instance->TXBAR_. The two functions are namely _HAL_FDCAN_AddMessageToTxFifoQ()_ , and _HAL_FDCAN_EnableTxBufferRequest()_

_HAL_FDCAN_AddMessageToTxFifoQ()_ seems to update only one FIFO and then transmit its data.

_HAL_FDCAN_EnableTxBufferRequest()_ seems to only update _hfdcan->Instance->TXBAR_

I am unable to find a function that calls _FDCAN_CopyMessageToRAM()_ multiple times to fill the _CAN TX FIFOs_ before updating _TXBAR_

Please help implement above request.

Thanks,
@rxa1031

rxa1031, 2021-06-09:
Hello,

Is it possible to change the _static_ typecast for _FDCAN_CopyMessageToRAM()_ and change it to just:

```void FDCAN_CopyMessageToRAM(FDCAN_HandleTypeDef *hfdcan, FDCAN_TxHeaderTypeDef *pTxHeader, uint8_t *pTxData, uint32_t BufferIndex)```

and extern it.

Please suggest.

Thanks,
@rxa1031

rxa1031, 2021-06-16:
Hello,

As the requested code piece / function was not available, I had to code a custom / my own function for this.

May I share the coded custom function here?

Regards,
@rxa1031

KORKAD, 2021-12-10:
Hello @rxa1031,

Thank you for your contribution. Your report will be forwarded to our development team for analyzis and if confirmed the fix will be deployed on new STM32Cube series under dev.
We will get back to you as soon as we have an answer.

Thanks in advence for your patience.

With regards,
KORKAD

ASELSTM, 2022-01-17:
ST Internal Reference: 121056

rxa1031, 2022-01-18:
Dear ST,

Attached is the modified CAN driver that I’ve been using.

Please refer to function CRC_Evans_FDCAN_AddMultipleMessagesToTxFifoQ() and suggest whether the implementation is correct.

Regards,
Rajeev

*External Message, L00k Carefully.*

#include "../inc/Includes.H"

#if (_DRA_ENABLE_HW_TEST_ == _DRA_ENABLE_CODE_)

//volatile uint32_t g_u32ThisPSCAliveCMI;
//volatile uint32_t g_ReadCAN_CMI;

# if defined(dTestOnlyCANRx)
uint32_t gu32CANRxCounts = 0;
uint32_t gu32RxFifo0ITs = 0;
# endif // of defined(dTestOnlyCANRx)

#define dDeclareTestCAN_Messages( CAN_Port )	\
	\
bool		bCAN##CAN_Port##IsPSCAliveMessageReceived = false;	\
uint8_t u8CAN##CAN_Port##CurrentMessageMakerValue = 0;	\
	\
/* PSC is Alive response header. */	\
const FDCAN_TxHeaderTypeDef gNoCAN##CAN_Port##PSCIsAliveResponseTxHeader =	\
{	\
	.Identifier = dCAN##CAN_Port##PSCIsAliveTxID,	/*!< Specifies the identifier.	\
																		 This parameter must be a number between:	\
																			- 0 and 0x7FF, if IdType is FDCAN_STANDARD_ID	\
																			- 0 and 0x1FFFFFFF, if IdType is FDCAN_EXTENDED_ID               */	\
	\
	.IdType = FDCAN_EXTENDED_ID,			/*!< Specifies the identifier type for the message that will be	\
																		 transmitted.	\
																		 This parameter can be a value of @ref FDCAN_id_type               */	\
	\
	.TxFrameType = FDCAN_DATA_FRAME,	/*!< Specifies the frame type of the message that will be transmitted.	\
																		 This parameter can be a value of @ref FDCAN_frame_type            */	\
	\
	.DataLength = dDummyDataLength,		/*!< Specifies the length of the frame that will be transmitted.	\
																			This parameter can be a value of @ref FDCAN_data_length_code     */	\
	\
	.ErrorStateIndicator = FDCAN_ESI_ACTIVE, /*!< Specifies the error state indicator.	\
																		 This parameter can be a value of @ref FDCAN_error_state_indicator */	\
	\
	.BitRateSwitch = FDCAN_BRS_OFF,		/*!< Specifies whether the Tx frame will be transmitted with or without	\
																		 bit rate switching.	\
																		 This parameter can be a value of @ref FDCAN_bit_rate_switching    */	\
	\
	.FDFormat = FDCAN_CLASSIC_CAN,		/*!< Specifies whether the Tx frame will be transmitted in classic or	\
																		 FD format.	\
																		 This parameter can be a value of @ref FDCAN_format                */	\
	\
	.TxEventFifoControl = dReqdCAN##CAN_Port##TxEventFIFOControlState,  /*!< Specifies the event FIFO control.	\
																		 This parameter can be a value of @ref FDCAN_EFC                   */	\
	\
	.MessageMarker = dDummyMessageMaker		/*!< Specifies the message marker to be copied into Tx Event FIFO	\
																		 element for identification of Tx message status.	\
																		 This parameter must be a number between 0 and 0xFF                */	\
	\
};

# if ( 1 == MX_FDCAN1 )

dDeclareTestCAN_Messages( 1 );

[... 1541 lines cut ...]
#    if (1 == dCAN2PSCLead)
		u32ThisPSCAliveCMI = dIsLeadPSCAliveCMI;
#    elif (1 == dCAN2PSCTrail)	// of (1 == dCAN2PSCLead)
		u32ThisPSCAliveCMI = dIsTrailPSCAliveCMI;
#    endif // of (1 == dCAN2PSCLead)
		__schedule_barrier();
		rvCAN_ReadIncomingFIFO(hfdcan, FDCAN_RX_FIFO0, (FDCAN_RxHeaderTypeDef *)&oCAN2RxHeader, au8CAN2Payload);
		__force_stores();
		__schedule_barrier();
#    if (_DRA_ENABLE_HW_TEST_ == _DRA_ENABLE_CODE_)
		if(!bCAN2IsPSCAliveMessageReceived && ( u32ThisPSCAliveCMI == MulCANMessageIDFromArbitration( oCAN2RxHeader.Identifier )))
		{
			bCAN2IsPSCAliveMessageReceived = true;
			__force_stores();
		}
#    endif // of (_DRA_ENABLE_HW_TEST_ == _DRA_ENABLE_CODE_)
		__schedule_barrier();
	}

#   endif // of (1 == MX_FDCAN2)
}

#  endif // of ( 1 == MX_FDCAN1 ) || ( 1 == MX_FDCAN2 )

#endif // of (_DRA_ENABLE_HW_TEST_ == _DRA_ENABLE_CODE_)

HBOSTM, 2023-08-01:
Hello @rxa1031,

Thank you for this report, currently there is no plan to implement this fix, as this type of modification may create a compatibility break in the existing applications. So please allow me to close this issue. 
Thank you for your comprehension.

Best Regards,

closed 2023-08-01 by HBOSTM